esphome/feature-requests #471: ESP8266 Arduino Core v2.6.0

OttoWinter, 2019-11-09:
Arduino Core v2.6.0 for ESP8266 was just released: https://github.com/esp8266/Arduino/releases/tag/2.6.0

There appear to be quite a few bugfixes that may help stability.

Some highlights:
 - [ ] i2c transatction fix: https://github.com/esp8266/Arduino/pull/6654
 - [ ] getCycleCount forced to be inline (means that micros() might have had an issue in ISR context before) - https://github.com/esp8266/Arduino/pull/6477
 - [ ] memory fence for xt_rsil (disabling interrupts) - https://github.com/esp8266/Arduino/pull/6301
   - [ ] Related: We should probably change the `disable_interrupts()`+`enable_interrupts()` function to be a scoped lock like `esp8266::InterruptLock`. Makes code cleaner
 - [ ] NONOS-SDK updated to 2.2.x, and appears to be the new default; TODO evaluate stability, check tools/platformio-build.py - https://github.com/esp8266/Arduino/pull/6257, https://github.com/esp8266/Arduino/pull/6272, https://github.com/esp8266/Arduino/pull/6724
 - [ ] NAT support; looks like wifi repeating is easy now; TODO evaluate stability and RAM usage - https://github.com/esp8266/Arduino/pull/6360
 - [ ] Fix for random LEAmDNS crash on startup - https://github.com/esp8266/Arduino/pull/6261
 - [ ] `ets_intr_lock` nested; from description appears to fix ESP8266 getting unreachable after a while - https://github.com/esp8266/Arduino/pull/6484, https://github.com/esp8266/Arduino/issues/2330
 - [ ] Add option for slower flash chips, behavior seems to kinda be similar to the infamous https://github.com/esphome/issues/issues/455; plus the fact that it depends on fw size would finally make sense; idk, don't want to get my hopes up too much yet - https://github.com/esp8266/Arduino/pull/6552

  - [ ] This appears to be an opt-in. How do we make this user-settable? Maybe we can do something like with flash mode: Manual upload is always with worst (dout) setting, but any subsequent OTA auto-overwrites the flash mode to the correct one.

Potential changes needed for:
 - [ ] toolchain update - docker images - https://github.com/esp8266/Arduino/pull/6273
 - [ ] timezone handling, upstream now has its own copy of IANA database. But with ESPHome that part is handled by python generator, maybe that messes with the timezone code. https://github.com/esp8266/Arduino/pull/6373
 - [ ] attachInterruptArg is now public with a changed signature, probably esphal.cpp needs to be updated. - https://github.com/esp8266/Arduino/pull/6048
 - [ ] LEAmDNS can handle multiple net interfaces now, we can remove the hack in util.cpp - https://github.com/esp8266/Arduino/pull/6224
 - [ ] PSTR macro updated, maybe needs some changes to logger - https://github.com/esp8266/Arduino/pull/6565

May want to copy some bugfixes downstream:
 - [ ] WiFi STA static IP fix, not sure what exactly/why yet - https://github.com/esp8266/Arduino/pull/6194
 - [ ] Changes to Software Serial - https://github.com/esp8266/Arduino/pull/6722/files

Other useful stuff:
 - [ ] `precache()` (like prefetch for instruction memory) - https://github.com/esp8266/Arduino/pull/6628
 - [ ] umm_malloc update - https://github.com/esp8266/Arduino/pull/6438

brandond, 2019-11-11:
esp8266/Arduino#6552 is actually super interesting if you read through the whole thread from the original issue in https://github.com/esp8266/Arduino/issues/6366.

The issue is not that low-end flash chips are slow - the issue is that cheap boards are poorly laid out, and at higher TX powers the RF interference actually causes the flash to misread, resulting in all kinds of bad things including executing instructions that send the pointer to bad offsets in the stack. I wonder if it might also be affecting writes and corrupting flash - that wasn't tested in the thread.

The fix was to either reduce TX power, use an external antenna, or reduce the flash frequency: https://github.com/esp8266/Arduino/issues/6366#issuecomment-525524150. ESPHome doesn't currently expose wifi power settings, but this might be a good reason to do so. See: https://github.com/esp8266/Arduino/issues/764#issuecomment-176125868

brandond, 2019-11-11:
https://github.com/esp8266/Arduino/pull/6674#issuecomment-548657008 is also interesting

Marlor, 2019-11-28:
Just a heads-up, Core 2.6.2 just got released and is in platform-espressif8266 2.3.1

closed 2019-12-04 by OttoWinter