esphome/feature-requests #473: Support for i2c sensor chirp! - the plant watering alarm

Destix, 2019-11-10:
**Describe the problem you have/What new integration you would like**

Please add support for i2c sensor:  chirp! - the plant watering alarm

description and code: https://wemakethings.net/chirp/
replica available on Aliexpress (CJMCU Plant https://www.aliexpress.com/af/CJMCU-Plant.html )
https://www.aliexpress.com/item/32912460636.html
https://www.aliexpress.com/item/32842054964.html

**Please describe your use case for this integration and alternatives you've tried:**

i2c capacitive humidity sensor for plants 

**Additional context**

Sample Arduino code

```
#include <Wire.h>

void writeI2CRegister8bit(int addr, int value) {
  Wire.beginTransmission(addr);
  Wire.write(value);
  Wire.endTransmission();
}

unsigned int readI2CRegister16bit(int addr, int reg) {
  Wire.beginTransmission(addr);
  Wire.write(reg);
  Wire.endTransmission();
  delay(20);
  Wire.requestFrom(addr, 2);
  unsigned int t = Wire.read() << 8;
  t = t | Wire.read();
  return t;
}

void setup() {
  Wire.begin();
  Serial.begin(9600);
  writeI2CRegister8bit(0x20, 6); //reset
}

void loop() {
  Serial.print(readI2CRegister16bit(0x20, 0)); //read capacitance register
  Serial.print(", ");
  Serial.print(readI2CRegister16bit(0x20, 5)); //temperature register
  Serial.print(", ");
  writeI2CRegister8bit(0x20, 3); //request light measurement 
  Serial.println(readI2CRegister16bit(0x20, 4)); //read light register
}
```

mredone, 2020-05-17:
Hello. How can i help add this sensor to esphome?

lupumihai, 2020-06-04:
Is there some update on this subject?

lupumihai, 2020-06-18:
I manage to make it work, even if I don't use all its functionality, I believe that my code could help you.
in my code I just read values from Chirp, in the same way, you could write registers to trigger light measurement or temperature by adding the "writeI2CRegister8bit( addr, value)" 

Setup for ESPHome is below + the attached extra code "chirp.h" - Remove the ".txt"

esphome:
  includes:
      - chirp.h

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: false

sensor:
 - platform: custom
    lambda: |-
      auto chirp = new Chirp();
      App.register_component(chirp);
      return {chirp -> humidity_RAW, chirp -> humidity, chirp -> calibStatus};
    sensors: 
    - name: ${upper_devicename}.Humidity_RAW
      unit_of_measurement: "LSB"
      accuracy_decimals: 0
      icon: mdi:water-percent 
      
    - name: ${upper_devicename}.Humidity
      unit_of_measurement: "%"
      accuracy_decimals: 0
      icon: mdi:water-percent 

    - name: ${upper_devicename}.Calib_Status
      icon: mdi:information-outline
      accuracy_decimals: 0
[chirp.h.txt](https://github.com/esphome/feature-requests/files/4798923/chirp.h.txt)

closed 2022-07-07 by nagyrobi