esphome/feature-requests #474: IO expander deferred register write

novirium, 2019-11-11:
**Describe the problem you have/What new integration you would like**

Currently doing intensive IO on a port expander (like the MCP23017) quickly clogs the I2C bus up, as there is an I2C register write for each individual `pin->digital_write()` call. Ideally there'd be some way to temporarily "defer" pin writes until you'd done the changes required - this would only be used by the IO expander components, and ignored by the normal native `digital_write`.

**Please describe your use case for this integration and alternatives you've tried:**

I'm currently trying to implement some support for stepper gauges (with an IO expander to handle multiple gauges), and part of that involves setting the 4 output pins at once for each step change. At the moment, this results in 4 I2C register transactions when there only really needs to be one. Speeding the bus up obviously helps a bit, but I was just wondering if anyone more familiar with ESPHome could point out a way of getting some sort of "hang on a sec till I've updated all my pin states" message to the IO expander components in a portable way? The brute force approach is to just add some global flag, turning it on while calling `digital_write` on your first 3 pins and turning it off on the last one. The IO expander component would then get patched to track "dirty" state in the pin registers, but that's pretty messy. Ideally the implementation would allow components needing fast simultaneous IO changes to wrap all but the last IO change in some sort of call or flag without needing to know if an IO expander is actually being used.

I'm hesitant to try and change the general esphal GPIOPin abstraction, but maybe adding an overloaded `virtual void digital_write(bool value, bool defer);` would work?

Or am I overthinking this, and things like the stepper component should be directly checking if the IO pin is on an expander, and directly talking to extra methods on that class?

**Additional context**

glmnet, 2019-11-12:
If it is the 23017 and you want to have all the esphome fanciness, you might add an api to the 23017 to write the full register at once, then use lambdas to calculate the register value and sent it at once. 
You’ll lose the digital outputs abstractions this way 
It’ll be hard to avoid lambdas, but you can also setup an action to use the ‘write_register’ more yamly 

Complex actions can be created so you can pass several parameters in a yaml action, each one being an output (bit)

OttoWinter, 2019-11-12:
So `GPIOPin` is primarily meant as an abstraction for native GPIO pins - using IO expanders was just a later addition because it fit in nicely.

I know of this problem and also thought of a similar solution. Ultimately though, that would mean all code that uses GPIO pins in some way would need to be aware of this. And I think that introduces a lot of unwanted complexity for integrations that just want to use gpio pins as in Arduino code.

Also there are tons of edge-cases, what should happen with native GPIO pins in deferred mode? Storing the deferred state there would just be a waste because they're written out sequentially anyway. What should happen with mixed native/expander set of pins? Also, GPIOPin code is highly optimized - any new feature such as deferral would hurt native pins performance which is not ok.

The solution is ultimately to create specialized integrations for the I/O expander like you said. Example: display.lcf_pcf8574.

closed 2019-11-12 by OttoWinter