esphome/feature-requests #476: Support for i2c FHT20 > SHT20 temp & humidity sensor

sibpph, 2019-11-12:
**Describe the problem you have/What new integration you would like**

Can you please help me integrate FHT20 sensor which uses SHT20 temp and humidity chip.
Product parameters:

Relative humidity:

Resolution: 12Bit

Repeatability: ±0.1%RH

Accuracy: 25 ° C ± 3% RH

Interchangeability: fully interchangeable

Response time: 1/e (63%) 25 ° C 8s

1m/s air 8s

Hysteresis: <±0.1%RH

Long-term stability: <±0.25% RH/yr

 

temperature:

Resolution: 14Bit

Repeatability: ±0.1 °C

Accuracy: 25 ° C ± 0.3 ° C

Response time: 1/e (63%) 10S

 

Electrical characteristics:

Power supply: DC 3.3 ~ 5.5V

Supply current: measuring 0.3mA standby 60μA

Sampling period: times greater than 2 seconds

 

Pin Description:

1, VDD power supply 3.3 ~ 5.5V DC

2, SDA bidirectional data line

3, GND ground, negative power supply

4, SCL clock line
**Please describe your use case for this integration and alternatives you've tried:**

I have tried platforms sht3xd and shtcx but i receive communication error 

[I][i2c:033]: Scanning i2c bus for active devices...
[07:55:56][I][i2c:040]: Found i2c device at address 0x40
[07:55:56][C][logger:175]: Logger:
[07:55:56][C][logger:176]:   Level: DEBUG
[07:55:56][C][logger:177]:   Log Baud Rate: 115200
[07:55:56][C][logger:178]:   Hardware UART: UART0
[07:55:56][C][status:034]: Status Binary Sensor 'Node01_living_parter Status'
[07:55:56][C][status:034]:   Device Class: 'connectivity'
[07:55:56][C][shtcx:055]: SHTCx:
[07:55:56][C][shtcx:056]:   Model: SHTC3
[07:55:56][C][shtcx:057]:   Address: 0x40
[07:55:56][E][shtcx:059]: Communication with SHTCx failed!
[07:55:56][C][shtcx:061]:   Update Interval: 60.0s
[07:55:56][C][shtcx:063]:   Temperature 'Temperatura Living Parter'
[07:55:56][C][shtcx:063]:     Unit of Measurement: '°C'
[07:55:56][C][shtcx:063]:     Accuracy Decimals: 1
[07:55:56][C][shtcx:063]:     Icon: 'mdi:thermometer'
[07:55:56][C][shtcx:064]:   Humidity 'Umiditate Living Parter'
[07:55:56][C][shtcx:064]:     Unit of Measurement: '%'
[07:55:56][C][shtcx:064]:     Accuracy Decimals: 1
[07:55:56][C][shtcx:064]:     Icon: 'mdi:water-percent'
**Additional context**

I found this for arduino:

#include <Wire.h>
#include "DFRobot_SHT20.h"
 
DFRobot_SHT20    sht20;
 
void setup()
{
    Serial.begin(9600);
    Serial.println("SHT20 Example!");
    sht20.initSHT20();                                  // Init SHT20 Sensor
    delay(100);
    sht20.checkSHT20();                                 // Check SHT20 Sensor
}
 
void loop()
{
    float humd = sht20.readHumidity();                  // Read Humidity
    float temp = sht20.readTemperature();               // Read Temperature
    Serial.print("Time:");
    Serial.print(millis());
    Serial.print(" Temperature:");
    Serial.print(temp, 1);
    Serial.print("C");
    Serial.print(" Humidity:");
    Serial.print(humd, 1);
    Serial.print("%");
    Serial.println();
    delay(1000);
}

torfbolt, 2019-11-14:
There's also a (vendor supplied) Linux kernel driver that covers the required functions to read SHT2x:
https://github.com/torvalds/linux/blob/master/drivers/hwmon/sht21.c

NicoFR75, 2019-11-18:
You can use custom sensor for reading SHT20, the "DFRobot_SHT20" library is working fine.

**YAML :**
```
esphome:
  name: <your_device_name>
  platform: ESP8266
  board: esp07
  board_flash_mode: dout
  includes:
    - <path_to_your_DFRobot_SHT20_folder>/DFRobot_SHT20.h
    - <path_to_your_DFRobot_SHT20_folder>/DFRobot_SHT20.cpp
    - SHT20.h

...

i2c:
  sda: GPIO13  <-- change if using different GPIO
  scl: GPIO12  <-- change if using different GPIO
  scan: true
  frequency: 400kHz

...

sensor:
  - platform: custom
    lambda: |-
      auto sht20 = new SHT20();
      App.register_component(sht20);
      return {sht20->temperature_sensor, sht20->humidity_sensor};
    sensors:
    - name: "Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 2
    - name: "Humidite"
      unit_of_measurement: "%"
      accuracy_decimals: 2
```

**SHT20.h (store it next to yaml file)**
```
#include "esphome.h"
#include "DFRobot_SHT20.h"

using namespace esphome;

class SHT20 : public PollingComponent {
 public:
  
  DFRobot_SHT20 myself;
  Sensor *temperature_sensor = new Sensor();
  Sensor *humidity_sensor = new Sensor();

  // Update every 5s (value in ms)
  SHT20() : PollingComponent(5000) { }

  void setup() override {
    myself.initSHT20();    // Init SHT20 Sensor
    delay(100);
    myself.checkSHT20();   // Check SHT20 Sensor
  }

  void update() override {
    float temperature = myself.readTemperature();
    temperature_sensor->publish_state(temperature);

    float humidity = myself.readHumidity();
    humidity_sensor->publish_state(humidity);
  }
};
```

JasonSwindle, 2020-09-30:
@NicoFR75 I tried your example, can I could not get it to work. Any more points of full examples to provide?

NicoFR75, 2020-09-30:
@JasonSwindle can you post your YAML file.

poldim, 2020-10-26:
@NicoFR75 Do you mind taking a look at https://github.com/poldim/esphome/tree/add-sht20/esphome/components/sht20
I'm trying to see if I can learn github enough to learn how to add it to the ESPHome core.  It's largely a copy of the SHT3X-D component but with some modifications I found from other libraries on github.

koenvervloesem, 2020-12-13:
@sibpph @JasonSwindle I just published an ESPHome configuration that uses the SHT20 sensor: https://github.com/koenvervloesem/M5Stack-Air-Quality-ESPHome.

Your `esphome:` section should have:

```yaml
  libraries:
    - "uFire SHT20"
    - "ArduinoJson"
  includes:
    - include/sht20.h
```

And then [include/sht20.h](https://github.com/koenvervloesem/M5Stack-Air-Quality-ESPHome/blob/main/include/sht20.h) is the custom component I made to support the sensor using the uFire SHT20 library.

Then add a custom sensor:

```yaml
sensor:
  - platform: custom
    lambda: |-
      auto sht20 = new SHT20();
      App.register_component(sht20);
      return {sht20->temperature_sensor, sht20->humidity_sensor, sht20->vpd_sensor, sht20->dew_point_sensor};
    sensors:
      - name: "Temperature"
        id: sensor_temperature
        unit_of_measurement: °C
        accuracy_decimals: 2
      - name: "Humidity"
        id: sensor_humidity
        unit_of_measurement: "%"
        accuracy_decimals: 2
      - name: "Vapour-pressure deficit"
        id: sensor_vpd
        unit_of_measurement: "kPa"
        accuracy_decimals: 2
      - name: "Dew point"
        id: sensor_dew_point
        unit_of_measurement: °C
        accuracy_decimals: 2
```

This is working fine here. An advantage of the uFire SHT20 library is that it also computes the vapour-pressure deficit and dew point based on its temperature and humidity. As you can see in the custom sensor above, I exposed these methods too in the custom component.

I haven't tried @poldim's component yet. I agree that adding this as a custom component to the ESPHome core would be the better solution, but I needed a quick solution to use this sensor at home.

closed 2022-07-01 by nagyrobi