esphome/feature-requests #477: Add Data support for Sim800L

carlos-sarmiento, 2019-11-13:
**Describe the problem you have/What new integration you would like**
I just saw that ESPHome added support for the Sim800L using SMS (#thanks @glmnet). Wanted to see if it would be possible to support HTTP GET/POST using [TinyGSM](https://github.com/vshymanskyy/TinyGSM) or another compatible library.

**Please describe your use case for this integration and alternatives you've tried:**
I would love to have IOT device in my car that can communicate with a cloud endpoint. 

**Additional context**

alexiri, 2020-10-22:
I'd also be interested in this, has there been any progress?

jperquin, 2020-12-09:
My use case is for a deployment of sensors and switches in a smart heating system for student houses where wifi coverage is usually spotty. Adding TinyGSM's functionality in addition to wifi comms would be very usefull.

Cellular coverage could serve either as backup or as primary source of internet connection.

The board we have in mind is the [https://www.amazon.nl/gp/product/B07VL9Z1SJ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1](https://www.amazon.nl/gp/product/B07VL9Z1SJ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

Messaging would go over MQTT so no need (from our side) for ESPHOME API..

netixx, 2021-01-05:
I have a use case for cellular connection as well. My intent is to run a remote node for watering plants in a remote location without Wifi connectivity. TiniGSM library seems to be compatible with pubsubclient at least, and could be a replacement for wifi connectivity.

I looked at [the documentation](https://esphome.io/custom/custom_component.html) but I don't think it could be used to provide network connectivity to other components.

To extend this idea further, the "cellular" device coult act as a gateway for other devices in the same location which wouldn't have cellular connection (using wifi AP mode, proxying messages/packets).

PiteousHonking, 2021-06-27:
I would also greatly appreciate this functionality to use mqtt via the SIM800L using an equivalent of the tiny gam library

cappern, 2022-01-05:
1+

MagicJF, 2023-01-14:
I would like to recieve video streams of my car dash cam into HA.
I guess support on this module might be a step into the right direction

PablexXXXX, 2023-11-20:
Off course, we need this. Imagine a node on any part of the plannet sending data to your home assistant instance using the wireguard component. JUST AWESOME.

oarcher, 2024-05-16:
I've submitted a draft PR at https://github.com/esphome/esphome/pull/6721 that enable TCP/IP for SIM7600  BG96 SIM7000  and SIM800. (Perhaps some others like SIM7070, etc..)

I've just tested It with [LilyGo T-SIM7600E](https://www.lilygo.cc/products/t-sim7600), and I need some alpha testers!

PablexXXXX, 2024-05-16:
> I've submitted a draft PR at esphome/esphome#6721

I will buy the module and test it. I will let ypu know how it goes

katsaplias, 2024-05-17:
just ordered this one
[https://www.banggood.com/LILYGO-T-A7608SA-H-T-A7608E-H-ESP32-SIM-LTE-Network-GPS-Antenna-ESP32-WROVER-E-Wireless-Module-WiFi-Bluetooth-Development-Board-p-1959313.html?rmmds=myorder&cur_warehouse=CN&ID=6320036](url)

closed 2026-10-23 by bdraco