esphome/feature-requests #478: Add Support for Gree AC / Heat Pump

icoma89, 2019-11-14:
**Describe the problem you have/What new integration you would like**

First and foremost, thanks for the wonderful system that esphome is. I would like to see support for the Gree AC protocol, similar to what is implemented in IRremoteESP8266 (https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Gree.cpp). I have used IRremoteESP8266 extensively in my projects and it works perfectly fine for Gree.

**Please describe your use case for this integration and alternatives you've tried:**
Direct implementation of the Gree protocol provides the same benefits that other direct implementations provide, no messing around with codes for users.

**Additional context**
I'm available to test if need be.

Gooman-rus, 2020-06-25:
Hello.
Is there a way to use this library (cpp) in esphome now?

11protos11, 2020-08-03:
I tried, but it didn't work.  Probably did something wrong.

vevsvevs, 2020-08-08:
I'm really interested in this too. Gree is the biggest Chinese OEM AC manufacturer with a very competitive quality/price, so their devices meet pretty commonly.

pcktdmp, 2020-08-18:
I got a working custom climate integration for Gree based on `IRremoteESP8266`, I'm in the process of prepping a PR to make it part of esphome.

younglip, 2020-08-28:
that's very important case for me. Please, add gree support

vevsvevs, 2020-09-03:
@pcktdmp thank you very much! Integration works perfectly, my AC is voice-controllable again 👍 It would be nice to have swing mode/fan speed as parameters, not as constants. And also receiver support (for the remote feedback) would be really nice too, but I think it relies on the climate_ir implementation?

pcktdmp, 2020-09-04:
@vevsvevs you are welcome, before creating a PR I'm aiming to use `climate_ir` instead of `climate` and more config options along the way doing that.

GT32b, 2020-09-27:
Thanks @pcktdmp, was waiting for this for a long while now. Any ETA integrate to main dev?

pcktdmp, 2020-10-07:
> Thanks @pcktdmp, was waiting for this for a long while now. Any ETA integrate to main dev?

I hope during the holidays. Unfortunately.

nithino, 2020-10-07:
How can i add gree custom component from the above repository to esphome in homeassistant? Please help.

martimarkov, 2020-12-25:
Hey, @pcktdmp are you working on this still or not?

Hagakurje, 2021-01-04:
There is another way to use IRremoteESP8266 with ESPHome:

1. Create file "gree_ir.h" in "config\esphome\gree", where "config" is HA configuration folder
gree_ir.h
```
#include "esphome.h"
#include "IRremoteESP8266.h"
#include "IRsend.h"
#include "ir_Gree.h"

const uint16_t kIrLed = 14; // transmitter GPIO pin to use
IRGreeAC ac(kIrLed);

class GreeAC : public Component, public Climate
{
public:
  void setup() override
  {
    ac.begin();
    ac.on();

    ac.setMode(kGreeAuto);
    ac.setTemp(22);
    ac.setFan(kGreeFanAuto);
    ac.setSwingVertical(true, kGreeSwingAuto);
    ac.setLight(false);
  }

  climate::ClimateTraits traits()
  {
    auto traits = climate::ClimateTraits();

    traits.set_supports_current_temperature(true);
    traits.set_supports_two_point_target_temperature(false);
    traits.set_visual_min_temperature(16);
    traits.set_visual_max_temperature(30);
    traits.set_visual_temperature_step(1.f);

    traits.set_supports_auto_mode(true);
    traits.set_supports_cool_mode(true);
    traits.set_supports_heat_mode(true);
    traits.set_supports_fan_only_mode(false);
    traits.set_supports_dry_mode(false);

    traits.set_supports_fan_mode_auto(true);
    traits.set_supports_fan_mode_low(true);
    traits.set_supports_fan_mode_medium(true);
    traits.set_supports_fan_mode_high(true);

    traits.set_supports_swing_mode_off(true);
    traits.set_supports_swing_mode_both(false);
    traits.set_supports_swing_mode_vertical(true);
    traits.set_supports_swing_mode_horizontal(false);

    return traits;
  }

  void control(const ClimateCall &call) override
  {
    if (call.get_mode().has_value())
    {
      ClimateMode climateMode = *call.get_mode();
      switch (climateMode)
      {
      case CLIMATE_MODE_HEAT:
        ac.setMode(kGreeHeat);
        ac.setLight(false);
        ac.on();
        break;
      case CLIMATE_MODE_COOL:
        ac.setMode(kGreeCool);
        ac.setLight(false);
        ac.on();
        break;
      case CLIMATE_MODE_AUTO:
        ac.setMode(kGreeAuto);
        ac.setLight(false);
        ac.on();
        break;
      case CLIMATE_MODE_OFF:
        ac.setLight(true);
        ac.off();
        break;
      }

      this->mode = climateMode;
      this->publish_state();
    }

    if (call.get_target_temperature().has_value())
    {
      float temp = *call.get_target_temperature();
      ac.setTemp(temp);

      this->target_temperature = temp;
      this->publish_state();
    }

    if (call.get_fan_mode().has_value())
    {
      ClimateFanMode fanMode = *call.get_fan_mode();
      switch (fanMode)
      {
      case CLIMATE_FAN_AUTO:
        ac.setFan(kGreeFanAuto);
        break;
      case CLIMATE_FAN_LOW:
        ac.setFan(kGreeFanMin);
        break;
      case CLIMATE_FAN_MEDIUM:
        ac.setFan(kGreeFanMed);
        break;
      case CLIMATE_FAN_HIGH:
        ac.setFan(kGreeFanMax);
        break;
      }

      this->fan_mode = fanMode;
      this->publish_state();
    }

    if (call.get_swing_mode().has_value())
    {
      ClimateSwingMode swingMode = *call.get_swing_mode();
      switch (swingMode)
      {
      case CLIMATE_SWING_OFF:
        ac.setSwingVertical(false, kGreeSwingMiddle);
        break;
      case CLIMATE_SWING_VERTICAL:
        ac.setSwingVertical(true, kGreeSwingAuto);
        break;
      }

      this->swing_mode = swingMode;
      this->publish_state();
    }

    ac.send();
    delay(200);
  }
};
```

Hagakurje, 2021-01-04:
There is another way to use IRremoteESP8266 with ESPHome:

1. Create file "gree_ir.h" in folder "config\esphome\gree\", where "config" is HA configuration folder 
gree_ir.h:
```
#include "esphome.h"
#include "IRremoteESP8266.h"
#include "IRsend.h"
#include "ir_Gree.h"

const uint16_t kIrLed = 14; // transmitter GPIO pin to use
IRGreeAC ac(kIrLed);

class GreeAC : public Component, public Climate
{
public:
  void setup() override
  {
    ac.begin();
    ac.on();

    ac.setMode(kGreeAuto);
    ac.setTemp(22);
    ac.setFan(kGreeFanAuto);
    ac.setSwingVertical(true, kGreeSwingAuto);
  }

  climate::ClimateTraits traits()
  {
    auto traits = climate::ClimateTraits();

    traits.set_supports_current_temperature(true);
    traits.set_supports_two_point_target_temperature(false);
    traits.set_visual_min_temperature(16);
    traits.set_visual_max_temperature(30);
    traits.set_visual_temperature_step(1.f);

    traits.set_supports_auto_mode(true);
    traits.set_supports_cool_mode(true);
    traits.set_supports_heat_mode(true);
    traits.set_supports_fan_only_mode(false);
    traits.set_supports_dry_mode(false);

    traits.set_supports_fan_mode_auto(true);
    traits.set_supports_fan_mode_low(true);
    traits.set_supports_fan_mode_medium(true);
    traits.set_supports_fan_mode_high(true);

    traits.set_supports_swing_mode_off(true);
    traits.set_supports_swing_mode_both(false);
    traits.set_supports_swing_mode_vertical(true);
    traits.set_supports_swing_mode_horizontal(false);

    return traits;
  }

  void control(const ClimateCall &call) override
  {
    if (call.get_mode().has_value())
    {
      ClimateMode climateMode = *call.get_mode();
      switch (climateMode)
      {
      case CLIMATE_MODE_HEAT:
        ac.setMode(kGreeHeat);
        ac.on();
        break;
      case CLIMATE_MODE_COOL:
        ac.setMode(kGreeCool);
        ac.on();
        break;
      case CLIMATE_MODE_AUTO:
        ac.setMode(kGreeAuto);
        ac.on();
        break;
      case CLIMATE_MODE_OFF:
        ac.off();
        break;
      }

      this->mode = climateMode;
      this->publish_state();
    }

    if (call.get_target_temperature().has_value())
    {
      float temp = *call.get_target_temperature();
      ac.setTemp(temp);

      this->target_temperature = temp;
      this->publish_state();
    }

    if (call.get_fan_mode().has_value())
    {
      ClimateFanMode fanMode = *call.get_fan_mode();
      switch (fanMode)
      {
      case CLIMATE_FAN_AUTO:
        ac.setFan(kGreeFanAuto);
        break;
      case CLIMATE_FAN_LOW:
        ac.setFan(kGreeFanMin);
        break;
      case CLIMATE_FAN_MEDIUM:
        ac.setFan(kGreeFanMed);
        break;
      case CLIMATE_FAN_HIGH:
        ac.setFan(kGreeFanMax);
        break;
      }

      this->fan_mode = fanMode;
      this->publish_state();
    }

    if (call.get_swing_mode().has_value())
    {
      ClimateSwingMode swingMode = *call.get_swing_mode();
      switch (swingMode)
      {
      case CLIMATE_SWING_OFF:
        ac.setSwingVertical(false, kGreeSwingMiddle);
        break;
      case CLIMATE_SWING_VERTICAL:
        ac.setSwingVertical(true, kGreeSwingAuto);
        break;
      }

      this->swing_mode = swingMode;
      this->publish_state();
    }

    ac.send();
    delay(200);
  }
};
```
2. Change transmitter pin in row "const uint16_t kIrLed = 14; // transmitter GPIO pin to use"
3. Modify yaml file:
```
...

esphome:
  name: ***
  platform: ***
  board: ***
  includes: 
    - gree/gree_ir.h
  libraries:
    - IRremoteESP8266

...
    
climate:
- platform: custom
  lambda: |-
    auto my_ac = new GreeAC();
    App.register_component(my_ac);
    return {my_ac};

  climates:
    - name: "My AC"
```
4. Enjoy :) 
[image]

pcktdmp, 2021-01-20:
> Hey, @pcktdmp are you working on this still or not?

Unfortunately not, @Hagakurje's solution would be the best option until I or someone else finds some time and priority to pick this up again.

Hagakurje, 2021-01-23:
updated version: https://github.com/Hagakurje/ESPHome_Gree_AC

Luth1ng, 2023-01-08:
Is anyone still working on this ?
@Hagakurje's solution is the better way to go at the moment, but a little tricky for new users.

I can be available for testing if needed.

yurhett, 2023-02-21:
I am trying to make changes to @Hagakurje's work to make it work better. I can be available for testing if needed.
If there are any ideas for native ESPHome support, I can assist in implementing them.

Hagakurje, 2023-03-04:
How about [arduino-heatpumpir](https://esphome.io/components/climate/climate_ir.html#arduino-heatpumpir) with gree or greeyac  protocol?
Does it work?

soloam, 2023-10-27:
This is a little forgotten on time! Hope we can bring it back! This is a very popular brand, it would be nice to have a native integration with ir transmitter and ir recover!

ayubn, 2023-11-17:
> There is another way to use IRremoteESP8266 with ESPHome:
>
> 1. Create file "gree_ir.h" in folder "config\esphome\gree", where "config" is HA configuration folder
>    gree_ir.h:
>
> ```
> #include [...]

thanks its working nice... i want to add default temperature settings how to do it

closed 2024-06-27 by nagyrobi