esphome/feature-requests #479: Add authentication for the web page

giacomofabbri, 2019-11-14:
**Describe the problem you have/What new integration you would like**

I would love to have an authentication for the access to the page!

**Please describe your use case for this integration and alternatives you've tried:**

I would love to have a subdomain on my domain that go directly to my internal IP, in this case I would be able to access ESPHome even from the outside using a password!

**Additional context**

rradar, 2019-11-17:
Looks like a duplicate of #274 to me

lwqcz, 2019-11-17:
It is solved - https://github.com/esphome/esphome/pull/674 nad in DOCs https://esphome.io/components/web_server.html

Therefore you can close this issue now, am I correct? 🤔 

BTW: I rather would never open port to ESP directly from the public Internet, it's not safe at all.

OttoWinter, 2019-11-17:
Correct, as @rradar and @lwqcz said this already exists.

I also agree with @lwqcz that exposing it publicly is not a good idea - there's a 99% chance that the ESPs TCP and HTTP stacks (external libraries) have some greater security vulnerability (the code has not gone through years of reviews like unix tools have)

closed 2019-11-17 by OttoWinter

giacomofabbri, 2019-11-19:
When I said to add the authentication I meant to add to the web interface for the ESPHome, http://localhost:6052 not to the single Wemos, etc..
I wanted to protect the main web interface
Please add it, it's pretty simple I think and could benefit a lot of people, thanks so much

giacomofabbri, 2019-11-19:
Solved! Thanks for everything