esphome/feature-requests #480: Add component to collect WIFI client signal strength

anhaehne, 2019-11-15:
**Describe the problem you have/What new integration you would like**

I would like the ESPHome to collect WIFI client signal strength readings and publish them to MQTT.

**Please describe your use case for this integration and alternatives you've tried:**

I'm currently working on an WIFI based indoor localization system. I would like to use ESPHome-Nodes as additional WIFI scan nodes to get more wifi client signal strength data.

Is this kind of WIFI usage possible in the current architecture? I would probably have turn of the existing WIFI connection, turn on AP mode and scan for a while before switching back. I wanted to make sure this idea is feasible before i started working on it.

**Additional context**

glmnet, 2019-11-15:
It will not fit ESPHome easily, once you collect the data you’ll have a list of ssid/ bssid and signal strength which you’ll want to store. You cannot setup sensors or another thing dynamically to store this. May be via mqtt you can figure something but that will not fit anything being used today. I don’t know if HA or something else can parse such data structure.
You can always use ESPHome as a base for some custom code which does almost anything, and take advantage of OTA, Logging, WiFi, etc.

anhaehne, 2019-11-15:
Could changing the wifi-mode somehow interfere with esphome working properly? Anyways i will try writing my own component to check how feasible my idea is. I think we can close this issue for now. Thanks for you help.

glmnet, 2019-11-15:
I really don't know if the ESP can survey the wifi networks without losing connection to your AP.
If you lose connection it will be harder because you'll end up with a lot of disconnection events. But should be doable.

Of course Wifi component in ESPHome is not meant to do that so you'll have to hack it.

Another option might be to have an slave ESP just doing wifi scans and providing results to main ESPHome node via serial. Just saying...

anhaehne, 2019-11-15:
Yes that would be the easy way. But since ESPHome is such a great platform i already have a bunch of them around the house.

glmnet, 2019-11-15:
In any case measuring client signal strength?? what client? You can measure AP's signal strengths only. AFAIK

anhaehne, 2019-11-15:
Something like this: https://www.hackster.io/rayburne/esp8266-mini-sniff-f6b93a

OttoWinter, 2019-11-16:
No, this is not going to work for ESPHome.

The way the presence detection you're speaking about works is by putting the device in promiscuous mode and matching the MAC address in the wifi packets. (Note: This is **not** AP mode, as initially suggested)

In principle, that's not a big problem, but: promiscuous mode and STA connection are mutually-exclusive. The ESP can't communicate with a WiFi AP through standard interfaces while in promiscuous mode.

The main reason to not allow this in ESPHome however is this: I don't see a way this will work well to the usual quality of ESPHome, and it would be a mess to maintain. There are already enough problems with WiFi and the ESPs, adding promiscuous mode on top of that is just asking for trouble.

Sure, in custom code (and potentially even a cookbook), this is fine - but I don't see a way that ESPHome can maintain that code because it's just going to be a mess and hard to maintain.

One thing can be added to the codebase though: API calls to the WiFiComponent class that suspend the WiFi connection process for a while (for example while scanning in promiscuous mode). That would not be that hard to add and doesn't add much maintenance cost.

closed 2019-11-16 by OttoWinter