esphome/feature-requests #481: Add support for ESP32 presence tracking with phones.

rarroyo6, 2019-11-15:
**Describe the problem you have/What new integration you would like**
Right now, the BLE tracker does not support devices which issue a random MAC address, such as a phone or tablet.

**Please describe your use case for this integration and alternatives you've tried:**
The built-in "bluetooth_tracker" in HA works properly, as long as the MAC address of the device is known.
It would be nice for the ESP32 tracker to be able to detect the devices if the MAC address is known, even if it broadcast a random address.
This way, multiple presence trackers can be placed throughout the hose, to avoid dead-spots.

**Additional context**
There is another project which does something similar here: [https://community.home-assistant.io/t/monitor-reliable-multi-user-distributed-bluetooth-occupancy-presence-detection/68505](https://community.home-assistant.io/t/monitor-reliable-multi-user-distributed-bluetooth-occupancy-presence-detection/68505)

austwhite, 2020-05-09:
Just wanted to watch this as I want this option as well :)

jaybinks, 2020-05-12:
watching

jlweston, 2020-06-06:
This would be amazing. I already have several ESP32s with ESPHome installed and BLE presence detection is the only thing they're missing. If the `monitor` project linked above can do it using `NAME` requests, would it be easily implemented in ESPHome?

xcray, 2020-07-07:
This feature request is opened on Nov 16 of 2019, and we already have a wonderful reference, so let's guess when could it become reality?

austwhite, 2020-07-07:
I hope this gets added soon.  I would love to use my existing ESP32's running ESPHome for lights to add some BLE tracking

DaAwesomeP, 2020-09-04:
It seems that this would require another separate BLE component in ESPHome. NAME requests should not be spammed to avoid causing too much 2.4 GHz traffic and to prevent the risk of missing a response. The flowcharts shown here should be implemented: https://community.home-assistant.io/t/monitor-reliable-multi-user-distributed-bluetooth-occupancy-presence-detection/68505

rarroyo6, 2021-09-07:
This would still be nice to have on esp-home, but I have not seen any new activity on this for over a year.
Is there anything on the plans, or should we consider this dead, and look for other solutions?

jbraceg, 2022-02-09:
> should we consider this dead, and look for other solutions?

Even if it isn't dead, given nothing has happened in more than a year I'd be interested in other solutions. Do you have something in mind that might potentially work?

DaAwesomeP, 2022-02-09:
It sounds like mayve nobody has worked on a solution, though this may still be the best solution (note I do not watch this repo so I am not aware of other issues).

The most ideal scenario is that the ESP would listen for new random MAC advertisements, asks if the known devices have arrived when there is such an advertisement, and then updates the state. When the advertisement expires, it should ask again.

See these two flowcharts from the post I linked to [above](https://github.com/esphome/feature-requests/issues/481#issuecomment-687246477):

[image]

However, I think this is maybe difficult to implement with only one 2.4GHz radio. Responding to the random MAC advertisements is hard if the radio can only be in Bluetooth or WiFi mode and not both simultaneously (I say this without knowledge of if the ESP32 can do this). You would miss advertisements while the device is in WiFi mode.

If you have to cycle between the modes, then the best you can do is this method (image also from post [above](https://github.com/esphome/feature-requests/issues/481#issuecomment-687246477):

[image]

However, this totally spams the 2.4 GHz band asking for names and has a (supposedly) noticeable effect for nearby WiFi devices if it is constant.

But a first step is definitely implementing the the time-interval-based scanning method.

myhomeiot, 2024-05-11:
> Right now, the BLE tracker does not support devices which issue a random MAC address, such as a phone or tablet.

Finally we get it https://github.com/esphome/feature-requests/issues/2285