esphome/feature-requests #482: WebSocket Client as alternative to MQTT

ellavas, 2019-11-16:
**Describe the problem you have/What new integration you would like**

I want to develop a websocket client to push meter consumption when a switch is on.

**Please describe your use case for this integration and alternatives you've tried:**

Currently, the only way to push data to a remote server is to use MQTT. This protocol is very nice and flexible but it requires the use of a MQTT broker. 
I want to integrate my switch with an IOT server which supports only connexion though websocket.
When the device is switched on, a session should start. Then, the consumption is sent every minutes. Finally, the session should be stopped when the device is switched off.
What is the best way to implement such feature ?

**Additional context**

lwqcz, 2019-11-17:
Did you consider using native API? https://esphome.io/components/api.html

ellavas, 2019-11-17:
I had a look to the native Api but this is not an option as I can't modify the backend and I don't want to add an additional server to act as a wrapper between the native API and the web socket backend.

OttoWinter, 2019-11-17:
What you're proposing is a generic websocket protocol - and from what I understand you want to control the format of those messages by the user. 

ESPHome can't create bases for something where the protocol is changeable by the user - as that would mean a 100% generic protocol.

You can certainly do this in custom code. You're free to write a component (like MQTT client or API server) that uses the objects from `App` and passes them into ws messages.

closed 2019-11-17 by OttoWinter

lwqcz, 2019-11-17:
So you need some kind of PUSH mechanism to transfer consumption data to IoT-server via specific WebSocket? 

I assume that something like https://esphome.io/components/text_sensor/custom.html would he a start, but anyway some Arduino IDE/C++ development will be needed if you do not want to have some kind of wrapper between those two devices.

ellavas, 2019-11-18:
Thanks for your replies, this is exactly what I have started to implement. 
A new custom component which setups a web socket connection and regularly sends the sensors status in the loop method.

@lwqcz : what is the purpose of a text sensor ? 

Additional question: Is it possible to call a method of my custom component from the yaml definition. For example, when a sensor value changed ?

OttoWinter, 2019-11-20:
> Additional question: Is it possible to call a method of my custom component from the yaml definition. For example, when a sensor value changed ?

You can hook into the sensor by attaching a callback with `add_on_state_callback` (see api docs).

If you want to hook into all sensors automagically, you can also use the `Controller` class.

ellavas, 2019-11-20:
Thanks a lot, I'll have a look