esphome/feature-requests #485: Comet Blue BLE valve support

lwqcz, 2019-11-17:
**Describe the problem you have/What new integration you would like**

A direct support for Comet Blue BLE Thermostat valves as binary sensor (I assume).

**Please describe your use case for this integration and alternatives you've tried:**

I'm using USB BT 4.0 dongle at my Pine64 to communicate with Comet Blue Thermostat BLE enabled valve controller on my heating radiators.
I would like to have a separate ESP32 to monitor and control these devices and there is even Arduino IDE based code to do so - https://github.com/MrKrabat/Comet-Blue-Thermostat
I would like to have direct support for this in ESPHome.

**Additional context**

Original product - https://eurotronic.org/produkte/bluetooth-heizkoerperthermostate/comet-blue/ 
Home-assistant.io discussion thread - https://community.home-assistant.io/t/eurotronic-cometblue-xavax-hama-sygonix/47783

GrumpyMeow, 2019-12-01:
I've started with creating a Comet Blue TRV climate-component.
The current state is only the plumbing: https://github.com/GrumpyMeow/esphome/tree/comet-blue
When i get around to completing it i will create a PR for it.

lwqcz, 2019-12-02:
@GrumpyMeow do you need to test it on Comet HW or not? :-) thanks anyway! 👍

GrumpyMeow, 2019-12-02:
Hi @lwqcz,
Unfortunatly i've ran into a dead-end with the implementation. There is currently no support for sending data to bluetooth devices in ESPHome yet. This needs to be added to ESPHome first, before the Comet Blue can be controlled through ESPHome.

I'm sincerly bummed out over not being able to complete this component as i love the user-friendlyness of ESPHome itself. 

For now.. I've paused my work on this component. I'll probably pick it up again in the future.

OttoWinter, 2019-12-04:
@GrumpyMeow Correct, BLE sending APIs (I'm assuming this means BLE GATT) is not supported - only BLE ADV is. But you can use the esp-idf APIs for that if needed.

lwqcz, 2019-12-04:
@OttoWinter Yeah, it is using BLE GATT.
@GrumpyMeow even ability to read data from valves would be better than nothing 👍

kerimt-dek, 2019-12-11:
If not possible by esphome itself, it must be possible to have esphome connected by I2C to another esp32 device programmed with Arduino IDE code @lwqcz provided

GrumpyMeow, 2020-01-11:
I recently gave it another try and seem to have made some progress. I found a project on github for controlling eq3 Bluetooth TRVs.

Apparently the eq3 uses ble-notifications to retrieve values, while the Comet Blue doesn’t. For this I had to extend the original project a bit. I think I’ll need support of the original developer to get my changes aligned. I’m have very limited CPP development experience.

There are still some other minor issues to iron-out. But I’m now actually able to connect, authenticatie and get/set temperature from Home Assistant. I’ll try to publish the component as soon as possible. I’m hoping that the original developer will adopt my Comet Blue implementation. 

When I make progress, I’ll post it here.

GrumpyMeow, 2020-01-12:
I committed my changes to my github repo: https://github.com/GrumpyMeow/esphome-components
Pincode is still hardcoded and needs to be 000000 on the device.

My configuration:
climate:
  - platform: cometblue
    pin: 0000
    id: cb2
    name: Comet Blue 2
    mac_address: ff:ff:ff:5a:3f:ff
    update_interval: 1min

danielkucera, 2020-02-16:
Unfortunately, I didn't know about progress with this device and created my own attempt:
https://github.com/danielkucera/esphome/tree/cometblue/esphome/components/cometblue

It is mostly working but has two big issues:
- doesn't work with multiple Cometblues (showstopper for me)
- uses `ESP32 BLE Arduino` which is the cause of the issue above

```
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Comet Blue Thermostat> 1.0.0 #f9caae5
|   |-- <ESP32 BLE Arduino> 1.0.1
|-- <Update> 1.0
|-- <WiFi> 1.0
```
@GrumpyMeow does your code work with multiple devices?

GrumpyMeow, 2020-02-19:
Hi Daniel,
I did manage to get everything working. And it does support multiple Comet Blue devices. I have it running for a while already in my own home for three CB devices.

The BLE range is very limited of the CB devices or Esp32. This is something to keep in mind.

The stability is something which I would want to improve. Because at times I see that no connection was made for hours and thus no data is received. At this time I don’t have any clue what the cause is. 

One thing I learned is to disconnect from the CB devices after the data has been transmitted.

danielkucera, 2020-02-20:
I am also not happy about the stability but it is shitty with any client software I tested (https://github.com/Hy3n4/ha-cc-cometblue, https://github.com/MrKrabat/Comet-Blue-Thermostat, https://github.com/xrucka/cometblue, ...)

So yesterday I opened the device but I got disappointed. 
https://blog.danman.eu/cometblue-teardown/
The BT stack is managed by DA14580 which has only one-time-programmable memory so there is no chance to improve the bluetooth software :(

lwqcz, 2020-06-18:
What about ESP8266/32 board instead original board + eink display? 🤔

TulkashAtomic, 2020-10-28:
> If not possible by esphome itself, it must be possible to have esphome connected by I2C to another esp32 device programmed with Arduino IDE code @lwqcz provided

Hi, I’m not sure if this is the right spot. Was unsure how to message @kerimt-dek  directly. I was looking for a way to control my Holman BTX Bluetooth tap timer with home assistant. I found this on GitHub which I think I can get working but not sure how to integrate it into home assistant. 

https://github.com/scottmckenzie/holman-linux-python

Not sure if it could be run on an esp32 though.  

Hope this helps somewhat. .

lwqcz, 2020-11-29:
@OttoWinter any chance of getting this sensor type into esphome? Anytime soon? Please let me know. I have valves and ESP8266+32 to test this new feature.

piwowarczykjacek, 2020-12-02:
@GrumpyMeow 
I installed cometblue library under esphome for homeassistant. unfortunately I can't find anywhere what the yaml project file should look like. I did my own but an error pops up. Can I ask for help. thank you
[image]

piwowarczykjacek, 2020-12-02:
[image]

I installed cometblue library under esphome for homeassistant. unfortunately I can't find anywhere what the yaml project file should look like. I did my own but an error pops up. Can I ask for help. thank you
[image]

closed 2022-07-01 by nagyrobi