esphome/feature-requests #487: WizNet W5500 ethernet board support

lwqcz, 2019-11-17:
**Describe the problem you have/What new integration you would like**
Direct support for Wiznet W5500 ethernet board, even for ESP8266 (as it lacks ethernet by design).

**Please describe your use case for this integration and alternatives you've tried:**
I would love to have ESP8266 with two W5500's as a bridge and wrapper between SNMP/XML only capable device and MQTT broker (and/or with HTTP PUSH service), even over wireless if useful. I have a lots of [GIOM 3000](http://www.elkoepusa.com/products/inels-bus-system/accessories-bus-system/meteostation-giom-3000-12065/) devices and it has no MQTT or PUSH capabilities useable over the Internet, so support for W5500 would be great.

**Additional context**
The chip used on the W5500 board does have datasheets [here]( https://cdn.sparkfun.com/datasheets/Dev/Arduino/Shields/W5500_datasheet_v1.0.2_1.pdf) and it is supported by [Arduino](https://www.arduino.cc/en/Reference/Ethernet) ([some blog post](https://leoricksimon.blogspot.com/2018/06/w5500-ethernet-module-with-arduino-mega.html) itself so implementation would be quite easy - but the main question is regarding memory consumption I assume.

lwqcz, 2019-11-17:
And some support for SNMP GET would be also interesting :-)

OttoWinter, 2019-11-20:
Nah, it's hard enough to support ESP8266 normal wifi stack, adding ethernet to that just adds too much complexity for a feature probably only few people would use.

If you really want ethernet, I'd suggest using an ESP32 - with the ESP32, the ethernet PHY is integrated well into the IP stack, and makes it easy for ESPHome to support.

For W5500 that is just too much work to add (and maintenance cost can also not be underestimated here).

closed 2019-11-20 by OttoWinter

lwqcz, 2019-11-22:
Thank you for considering this option and I tend to agree, its too much work for too little profit.