esphome/feature-requests #488: WH1080 and similar support

lwqcz, 2019-11-17:
**Describe the problem you have/What new integration you would like**
Implement support for cheap weather station HW like WH1080 and similar is.

**Please describe your use case for this integration and alternatives you've tried:**
Hardware like WH1080 is cheap and easy to implement but there is no direct support in ESPHome and it would be interesting for rather many people.

**Additional context**
Actual Arduino library is made by @pilotak available [here](https://github.com/pilotak/WeatherMeters).

pilotak, 2019-11-17:
I actually have already a working code https://github.com/pilotak/WemosWeatherStation but why not have it as part of esphome

lwqcz, 2019-11-17:
Yeah, I've noticed that :-) Your work is really impressive and I assume that as part of ESPHome it could be in many cases even better :-)

pilotak, 2019-11-17:
that's true, "Děkuji za pochvalu"