esphome/feature-requests #490: Improving PMSX003 component

guffy1234, 2019-11-18:
**Describe the problem you have/What new integration you would like**

hi

Yesterday I have tested my prototype of air quality station with pms7003 and found that PMSX003 reports data too frequently. As workaround i had added median and throttle filters. 
Also i have found that lifespan of the sensor in continuous mode isn't too large
8000h according this page https://www.b-io.info/posts/studies/dust_sensor/
and 3 yeas according an alibaba page https://www.alibaba.com/product-detail/PMS7003-plantower-high-quality-laser-pm2_60704594861.html

I saw this issue where you suggest to use genetic GPIO to enforce  the module to sleep. https://github.com/esphome/feature-requests/issues/269 

It's much better then nothing ) But i think that component could implement more sophisticated approach. 

**Please describe your use case for this integration and alternatives you've tried:**

As example it can use TX line to control the module with commands and turn it to passive mode. After switch to passive mode PMSX003  could be implemented as PolllingComponent and ask the sensor periadically (like mh-z19 does).
Also it could enforce sleep mode or wake the device up via serial command. No need to use SET wire for this.

**Additional context**

And, may be, it would be good to have ability to specify RESET pin to reset it on boot. Just for a case when firmware of PMS sensor hangs and you would reset it remotely by restart of the ESPHome device without turn off//on power.

guffy1234, 2019-11-18:
https://github.com/MartyMacGyver/PMS7003-on-Particle/blob/master/PMS7003_datasheet.pdf

redstreet, 2020-11-15:
Started [this thread](https://community.home-assistant.io/t/ignoring-sensor-values-during-warm-up-after-periodic-wake-up/245401).

redstreet, 2020-11-15:
> As example it can use TX line to control the module with commands and turn it to passive mode. 

In my understanding, this won't help sensor life, since the laser and fan are on fully in passive mode. It's just that data is not being transmitted.