esphome/feature-requests #491: Support AHT10 (I2C)

Joshfindit, 2019-11-18:
**Describe the problem you have/What new integration you would like**

Be able to read the AHT10

**Please describe your use case for this integration and alternatives you've tried:**

Debug logs show the device
```
Nov 18 12:48:41 d3miniv3breadboarder/debug [I][i2c:033]: Scanning i2c bus for active devices...
Nov 18 12:48:41 d3miniv3breadboarder/debug [I][i2c:040]: Found i2c device at address 0x38
```
I tried researching the documentation/feature requests, but could not find anything

**Additional context**

Arduino code: https://github.com/Thinary/AHT10/blob/master/src/Thinary_AHT10.cpp

gmasse, 2020-01-08:
Hi @Joshfindit,
You can try my porting: https://github.com/gmasse/esphome/tree/aht10-temperature-humidity/esphome/components/aht10
I have an issue to get Humidity (always returning 0). I suspect the hardware is buggy.
Waiting for new components to confirm and create the PR.

muxa, 2020-07-12:
I see the PR for this feature is merged, but this ticket is still open and `aht10` platform is not yet supported in lasted esphome. When is this expected to be released?

muxa, 2020-08-20:
@AndroVet it's in 1.15 release (currently in beta): https://github.com/esphome/esphome/releases

AndroVet, 2020-08-20:
Could you show me please what should I do with the zip-archive I downloaded?

AndroVet, 2020-08-21:
Help please, otherwise I really don't want to spend a more expensive BME680 sensor on air conditioning
[image]

AndroVet, 2020-08-27:
> @AndroVet it's in 1.15 release (currently in beta): https://github.com/esphome/esphome/releases

Could you show me please what should I do with the zip-archive I downloaded?

sumirati, 2022-05-15:
This feature request can be closed.

closed 2022-07-01 by nagyrobi