esphome/feature-requests #492: If possible: Generic i2c support

Joshfindit, 2019-11-18:
**Describe the problem you have/What new integration you would like**

- [x] To be able to scan the i2c bus and find the connected device
- [ ] To be able to specify a `generic` "platform" and scan the device for values
- [ ] To be able to give a name, ID, and sensor filters to each of those values

**Please describe your use case for this integration and alternatives you've tried:**

To easy fiddle around with random i2c chips

**Additional context**

I did look at https://esphome.io/custom/i2c.html and it *does* look like a way to accomplish this, but it's daunting and seems very complicated to do from the homeassistant integration

Fair warning: this is literally my first experience with i2c so I have no idea whether this is actually possible.

brandond, 2019-11-18:
i2c is a bus that devices can use to communicate. There's no standard for exposing things like sensor values. Each chip usually needs to be run through a setup sequence to configure it properly by sending commands to set memory registers, after which you can read other memory registers and convert those to meaningful values. 

Doing all of this requires reading datasheets, using other code as an example, and wrapping it in ESPHome's component+sensor API. That usually requires a fair bit of code, and isn't something you can just toss into a couple lines of YAML. ESPHome gives you the ability to do this with a custom component (as you linked).

Joshfindit, 2019-11-19:
@brandond As the 7-bit address is already known, would it not be a mater of reding whatever data is at that address?
I’m sure there are all sorts of odd edge cases, but I’m trying to wrap my head around how sensor data would be designed to be complicated to read.

Assuming this is accurate:

[image]

It seems to my layman’s eye that you could send a request to the known address, capture all data frames, and then count that as an unknown value. Send that value to a sensor filter and you can do whatever math is needed to get the real value.

If there’s such a thing as master sending an initial “send me x back” data frame, then maybe that could even be a parameter.

glmnet, 2019-11-19:
It’s just a small step from there to writing a custom i2c component and another small one to do a supported esphome integration. 
All basic boilerplate code to interact with i2c is already done and creating an additional i2c sensor is quite simple. What’s left is understanding the device inner workings, as @brandond said. 
Ideally every hardware i2c device would have a supported esphome integration so everyone can **quickly** benefit from it. 

Esphome yaml is not a general purpose language. It leverages lambdas to do many things that yaml is not the best choice. 

While what you request is doable, I doubt it will be the preferred way to do it for someone who can actually write both yaml and (a bit of) c++

And this is one of the things that makes esphome great, I can write c++ but I’d rather do yaml for gluing components together and small Automations.

Joshfindit, 2019-11-19:
Yep, after this conversation I started looking at datasheets for a couple different I2C chips, and it's 1000% not as straightforward as I imagined. To that end, I'm closing this as not realistic.
Thanks and cheers to @brandond and @glmnet

closed 2019-11-19 by Joshfindit