esphome/feature-requests #493: Support HDC2080 (I2C)

Joshfindit, 2019-11-19:
**Describe the problem you have/What new integration you would like**

Support for the HDC2080 temperature sensor

**Please describe your use case for this integration and alternatives you've tried:**

Standard stuff: want an esphome-based thermostat

_Side note: There are a *lot* of temp/humidity sensors available. If someone has the raw data and I have dev blessing, I'll build a documentation page about the best options when using ESPHome: what's used and praised by maintainers, relative accuracy, what the trade-offs are, etc. Possibly titled "Measuring indoor/outdoor temperature with ESPHome"_

**Additional context**

Test results showing it's very accurate for humidity, and a couple degrees off for temperature:
https://wiki.liutyi.info/display/ARDUINO/Test+4+v6+Saturated+NaCl+solution
Example code: https://github.com/tinkeringtech/HDC2080_breakout
Datasheet: http://www.ti.com/lit/ds/symlink/hdc2080.pdf
Application notes etc: http://www.ti.com/product/HDC2080/technicaldocuments

brandond, 2019-11-19:
You might look through the options on https://esphome.io/ in particular:
https://esphome.io/components/sensor/dht.html.
https://esphome.io/components/sensor/htu21d.html

Some of the particulate and gas sensors also provide temp and humidity:
https://esphome.io/components/sensor/pmsx003.html
https://esphome.io/components/sensor/scd30.html

ferazambuja, 2019-11-24:
It would be great to have a sensor comparison material! 
I saw that your example has a lot of sensors that are already supported, I would be great if you could contribute.

The HDC2080 might look similar to the HDC1080, have you tried?

Joshfindit, 2019-11-25:
> The HDC2080 might look similar to the HDC1080, have you tried?

I did, yes. Unfortunately it's too different 😔
```
Nov 25 15:02:43 d3miniv3breadboarder/debug [I][i2c:033]: Scanning i2c bus for active devices...
Nov 25 15:02:43 d3miniv3breadboarder/debug [I][i2c:040]: Found i2c device at address 0x40
Nov 25 15:02:43 d3miniv3breadboarder/debug [C][logger:175]: Logger:
Nov 25 15:02:43 d3miniv3breadboarder/debug [C][logger:176]:   Level: DEBUG
Nov 25 15:02:43 d3miniv3breadboarder/debug [C][logger:177]:   Log Baud Rate: 115200
Nov 25 15:02:43 d3miniv3breadboarder/debug [C][logger:178]:   Hardware UART: UART0
Nov 25 15:02:43 d3miniv3breadboarder/debug [C][hdc1080:028]: HDC1080:
Nov 25 15:02:43 d3miniv3breadboarder/debug [C][hdc1080:029]:   Address: 0x40
Nov 25 15:02:43 d3miniv3breadboarder/debug [E][hdc1080:031]: Communication with HDC1080 failed!
```

valordk, 2019-11-30:
I'm mostly familiar with Sensirions environmental sensors and the popular models are already implemented: 
[STS3x: Temp](https://esphome.io/components/sensor/sts3x.html)
[SHTCx: Temp + Humidity](https://esphome.io/components/sensor/shtcx.html)
[SHTx-D: Temp + Humidity](https://esphome.io/components/sensor/sht3xd.html)
[SCD30: CO₂ + Temp + Humidity](https://esphome.io/components/sensor/scd30.html)
[SGP30: CO₂eq + VOC](https://esphome.io/components/sensor/sgp30.html)

The SPS30: Particulate Matter support is also coming

For Temp & Humidity sensors there is detailed table which might be useful
[image]

regards,
Nad

NODeeJay, 2023-06-09:
I tried the same, but I get a different error, and I actually get reportings, but always -40 and 0, so there seem to be communication over the bus
`[D][hdc1080:069]: Got temperature=-40.0°C humidity=0.0%`

```
[C][hdc1080:016]: Setting up HDC1080...
[W][hdc1080:025]: HDC1080 initial config instruction error
[...]
[C][i2c.arduino:053]: I2C Bus:
[C][i2c.arduino:054]:   SDA Pin: GPIO21
[C][i2c.arduino:055]:   SCL Pin: GPIO22
[C][i2c.arduino:056]:   Frequency: 100000 Hz
[C][i2c.arduino:059]:   Recovery: bus successfully recovered
[I][i2c.arduino:069]: Results from i2c bus scan:
[I][i2c.arduino:075]: Found i2c device at address 0x40
```

[here I read ](https://e2e.ti.com/support/sensors-group/sensors/f/sensors-forum/861532/hdc2010-technical-differences-among-hdc2080-hdc2010-hdc1080)the sensor supports i2c with either 100k or 400k data rate, but setting i2c frequency to either 100000 or 400000 did not made it work either

ptr727, 2024-08-08:
+1 for HDC3020, much more RH (0.5%) accurate compared to e.g. ScioSense ENS212 (1.5%) or Sensirion SHT41 (2%) for about the same price.

NODeeJay, 2024-08-08:
> +1 for HDC3020, much more RH (0.5%) accurate compared to e.g. ScioSense ENS212 (1.5%) or Sensirion SHT41 (2%) for about the same price.

Can you back this with some test results?

ptr727, 2024-08-08:
> Can you back this with some test results?

No, per spec sheets:

- [ScioSense ENS212](https://www.sciosense.com/ens21x-family-of-high-performance-digital-temperature-and-humidity-sensors/), ±0.15°C, ±1.5%, 3s.
- [Sensirion SHT41](https://sensirion.com/search/products?q=SHT4x), ±0.3°C, ±2%, 4s.
- [TI HDC3020](https://www.ti.com/product/HDC3020), ±0.1°C, ±0.5%, 4s.

valordk, 2024-08-08:
TI HDC302x looks very attractive as it offers four selectable I2C addresses (0x44, 0x45, 0x46, 0x47)

Server8U2, 2025-02-21:
I have been evaluating the Adafruit HDC3022.  Hands down, it's the best temp sensor I've ever tested.
Now, I'm sad there is no support in ESPHome so I'm trying to learn how to add it myself.

towerclimber87, 2025-04-04:
#Server8U2, have you had any luck getting it to work?

PaulyMooch89, 2025-05-20:
+1 on the HDC3022 here, just picked up a few of them for some projects.  any luck @Server8U2 or @towerclimber87 ?

towerclimber87, 2025-05-21:
I’ve had no luck yet.

G-Pereira, 2025-07-05:
I've got it working al filled a PR: https://github.com/esphome/esphome/pull/9331

redstone99, 2025-08-11:
I filed PR supporting hdc302x: esphome/esphome#10187

G-Pereira, 2025-08-20:
They are still widely used and sold, even the much older brother HDC1080 which ESPHome supports.
I haven't had the opportunity to return to this but I will!