esphome/feature-requests #494: Support for Sensirion SHT1x sensors

m-pavel, 2019-11-20:
**Describe the problem you have/What new integration you would like**

Support for the Sensirion SHT1x temperature/humidity sensors
**Please describe your use case for this integration and alternatives you've tried:**

Standard stuff: read data, post to MQTT.
**Additional context**

m-pavel, 2019-11-22:
It's not exact I2C device, I did custom sensor for SHT1x using [this](https://github.com/practicalarduino/SHT1x) library. Here is a gist
 https://gist.github.com/m-pavel/14a03c967067ee8a8b11fa34184fe3e2

Nzxtime, 2020-09-01:
How did you connect the sensor with your ESP? I tried your solution and somehow I don't get any data or -40 °C when nothing is connected.

m-pavel, 2020-09-03:
Nothing special, just 10k pullup resistor between Vcc and data pin.

Nzxtime, 2020-09-04:
So I tried to connect every pin accordingly to the pinout (see below) and put a 10k pullup resistor between Vcc und data pin, but I still can't get any information in ESPHome.

Thank you in advance. 

[image]

nagyrobi, 2022-07-01:
SHT1x is [end of life](https://camo.githubusercontent.com/b78a40573d3b541902191a921d7f94798095a3b5b025294de13609880832b525/68747470733a2f2f692e696d6775722e636f6d2f4e4e654c73366a2e6a7067)...

landonr, 2024-01-30:
https://github.com/landonr/esphome-components/tree/main/components/SHT1XSensor

```yaml
external_components:
  - source: github://landonr/esphome-components
    components: [ SHT1XSensor ]

SHT1XSensor:
  update_interval: 10s
  data_pin: 22
  clock_pin: 23
  humidity: 
    id: humidity_sensor
    internal: false
  temperature: 
    id: temperature_sensor
    internal: false
    
```