esphome/feature-requests #496: Support for GDK101 radon sensor

minsuke, 2019-11-21:
**Describe the problem you have/What new integration you would like**

I got GDK101 Sensor, detects the radon 1m and 10m 
I used the Custom I2c and I am not developer but I want to support ESPHOME 

**Please describe your use case for this integration and alternatives you've tried:**

#include "esphome.h"

class MyCustomComponent : public Component {
 public:
  void setup() override {
    // Initialize the device here. Usually Wire.begin() will be called in here,
    // though that call is unnecessary if you have an 'i2c:' entry in your config

    // Wire.begin();
  }
  void loop() override {
    // Example: write the value 0x42 to register 0x78 of device with address 0x21
    Wire.beginTransmission(0x18);
    float radon1 = Wire.read(0xB4);
    radon1_sensor->publish_state(radon1);

    float radon10 = Wire.read(0xA0);
    radon10_sensor->publish_state(radon10);
    Wire.endTransmission();
  }
};

I wrote some codes esphome.io Custom I2C components but I failed. 

**Additional context**

https://www.eleparts.co.kr/data/goods_old/design/product_file/Hoon/AN_GDK101_V1.0_UART.pdf
https://www.eleparts.co.kr/data/goods_old/design/product_file/Hoon/GDK101datasheet_v1.5.pdf

I think this can be supported C++ develpers, I wish I can integrate this sensor.
This radone sensor maybe many people want to integrate their Air condition Sensor. 

I wish many people can detects their Enviroment with ESPOMHE. 

Thx again ESPHOME makers plz help me use my radon sensor. :)

minsuke, 2019-11-21:
and this is my esphome.yaml file 

  - platform: custom
    lambda: |-
      auto radone = new MyCustomComponent();
      App.register_component(radone);
      return {radone->radon1_sensor, radone->radon10_sensor};
      
    sensors:
      - name: "${upper_devicename} radone10 Sensor"
      
      - name: "${upper_devicename} radone1 Sensor"

minsuke, 2019-11-21:
This is Error codes when I compile 

In file included from src/main.cpp:155:0:
src/esphome_i2c.h: In member function 'virtual void MyCustomComponent::loop()':
src/esphome_i2c.h:14:34: error: no matching function for call to 'TwoWire::read(int)'
     float radon1 = Wire.read(0xB4);
                                  ^
In file included from src/esphome/components/i2c/i2c.h:3:0,
                 from src/esphome/components/bme680/bme680.h:5,
                 from src/esphome.h:13,
                 from src/main.cpp:3:
/root/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:104:9: note: candidate: virtual int TwoWire::read()
     int read(void);
         ^
/root/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:104:9: note:   candidate expects 0 arguments, 1 provided
In file included from src/main.cpp:155:0:
src/esphome_i2c.h:15:5: error: 'radon1_sensor' was not declared in this scope
     radon1_sensor->publish_state(radon1);
     ^
src/esphome_i2c.h:17:35: error: no matching function for call to 'TwoWire::read(int)'
     float radon10 = Wire.read(0xA0);
                                   ^
In file included from src/esphome/components/i2c/i2c.h:3:0,
                 from src/esphome/components/bme680/bme680.h:5,
                 from src/esphome.h:13,
                 from src/main.cpp:3:
/root/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:104:9: note: candidate: virtual int TwoWire::read()
     int read(void);
         ^
/root/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:104:9: note:   candidate expects 0 arguments, 1 provided
In file included from src/main.cpp:155:0:
src/esphome_i2c.h:18:5: error: 'radon10_sensor' was not declared in this scope
     radon10_sensor->publish_state(radon10);
     ^
src/main.cpp: In lambda function:
src/main.cpp:1141:23: error: 'class MyCustomComponent' has no member named 'radon1_sensor'
       return {radone->radon1_sensor, radone->radon10_sensor};
                       ^
src/main.cpp:1141:46: error: 'class MyCustomComponent' has no member named 'radon10_sensor'
       return {radone->radon1_sensor, radone->radon10_sensor};
                                              ^
src/main.cpp:1141:60: error: could not convert '{<expression error>, <expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::sensor::Sensor*>'
       return {radone->radon1_sensor, radone->radon10_sensor};
                                                            ^
Compiling /data/kitchen_sensors/.pioenvs/kitchen_sensors/lib789/WiFi/WiFi.cpp.o
*** [/data/kitchen_sensors/.pioenvs/kitchen_sensors/src/main.cpp.o] Error 1

glmnet, 2019-11-22:
Well, there are several issues here.

First, read does not receive a value, e.g. `Wire.read(value)` you can only do `Wire.read()`, this will return a value, what you get depends on what you've send before (with write)

Also, read is not going to return a float, it will return int, then you'll have to convert it to a float may be.

Also your sensor is not declared, thats why there are many `^` arrows pointing to `radonX_sensor` you must follow the [guide](https://esphome.io/components/sensor/custom.html#bonus-sensors-with-multiple-output-values)
you are missing the lines..

````
Sensor *radon1_sensor ....
Sensor *radon10_sensor ....
````

minsuke, 2019-11-22:
> Well, there are several issues here.
>
> First, read does not receive a value, e.g. `Wire.read(value)` you can only do `Wire.read()`, this will return a value, what you get depends on what you've send [...]

thx for your kind reply 
but I think I should learn C++ 

so If you can make me radone custom codes, I am very happy with that 

I'll contribute to another way to spread ESPHOME platform
ESPHOME is amazing platform to integrate Home IOT Devices. 
so I introduce ESPHOME codes to my country everyday when new amazing things can be made. 

I always grateful to your codes and many amazing iot things code. 

thx again : )

nagyrobi, 2022-07-01:
Please contact on Discord server to get support.

closed 2022-07-01 by nagyrobi