esphome/feature-requests #497: Support for Tuya v0 (battery powered devices)

DeadEnded, 2019-11-21:
**Describe the problem you have/What new integration you would like**

The current Tuya component is designed around protocol version 3.  Battery powered devices appear to use protocol version 0.  Adding the ability to use both will allow battery powered Tuya devices to be added (door sensors, PIR sensors, etc.)

**Please describe your use case for this integration and alternatives you've tried:**

The use case would be for battery powered tuya devices.  I tried other firmware but since the Tuya devices use an MCU the other firmware didn't support them... plus I really like ESPHome :-).

**Additional context**

It appears the Tuya component is setup using the protocol version 3 - the command type is as follows:
```
enum class TuyaCommandType : uint8_t {
  HEARTBEAT = 0x00,
  QUERY_PRODUCT = 0x01,
  MCU_CONF = 0x02,
  WIFI_STATE = 0x03,
  WIFI_RESET = 0x04,
  WIFI_SELECT = 0x05,
  SET_DATAPOINT = 0x06,
  STATE = 0x07,
  QUERY_STATE = 0x08,
};
```

But the window/door sensor appears to use version 0, the command comes through with the datapoint on command type 0x05... so the current Tuya component would see that as a WIFI select instead of the DATAPOINT.

I was able to find some useful information on this as follows:

Tuya component command enum in ESPHome code:
https://github.com/esphome/esphome/blob/dev/esphome/components/tuya/tuya.h#L35

Matching TUYA protocol (v3) information:
https://docs.tuya.com/docDetail?code=K8uhkesnz5mie

Information from testing by @Brandond  who had comments in the Tuya feature request (his was the second comment).  He has actually created a custom component before Tuya was officially supported.
https://github.com/esphome/feature-requests/issues/352#issuecomment-0517609285

@Brandond did testing on a door sensor and PIR sensor... although it is a different model than I am using, it uses the same protocol.  I have tested and confirmed my device uses the same messages.
https://github.com/brandond/esphome-tuya_pir#data-point-messages

At this point I would love to help develop the PR to update the Tuya component to be able to have a version selection, and then also add a new component for the door sensor... but I just do not have the skill or knowledge necessary.  I am happy to assist as much as possible (and to learn as much as well!) with getting this developed.

Thank you!
DeadEnd

brandond, 2019-11-21:
I haven't managed to get it merged in as an in-tree component yet, but I've been using this successfully for a few months: https://github.com/brandond/esphome-tuya_pir

See also https://github.com/esphome/esphome/issues/306

DeadEnded, 2019-11-21:
Thanks!
I am no software engineer, just a tinkerer who tries to glean knowledge for those around him :-).
If there is anything I can do to help please let me know - I was initially thinking I could do more, but I'm just to inexperienced at this point.

Since you have already done a MASSIVE amount of work on this (and it is nearly done), I will leave it to you and OttoWinter to make the decisions.

One question I do have... once this is merged, will it work with the Home Assistant API?
I believe your current custom component only uses MQTT, but the discussion on 306 was many months ago - so maybe it already would work?

Thanks!
DeadEnd

brandond, 2019-11-21:
I won't work as long as the API requires the HA server to connect to the device. The device is only online for a very short amount of time; it can't sit around for almost a minute waiting for the API client to connect.

rradar, 2019-12-14:
I think a good alternative to mqtt (while waiting to get the api ready to handle a push) would to let the device call a **http_request** (https://github.com/esphome/esphome/pull/719) to trigger a home assistant webhook for example.

Would be very nice to get this component 'ready' for this option ;)

nvx, 2022-01-29:
Worth noting Tasmota has support for this with the "TuyaMCU 51,20" option, for example for this device: https://templates.blakadder.com/mirabella_genio_I002576.html

vitalykorolev, 2022-02-03:
I have a number of reed switches that currently use @brandond implementation, but would love to see an official integration. I too could try to help with development and testing. I just don't have experience developing esphome integrations.

nvx, 2022-07-02:
Is this the Tuya Select feature that got added in 2022.6.0, or something else? Will have to try it out shortly!

nagyrobi, 2022-07-02:
@brandond