esphome/feature-requests #499: TTGO-T-Display board definition

KamilLenica, 2019-11-23:
**Describe the problem you have/What new integration you would like**

Add definition for TTGO-T-Display ESP32 board

**Please describe your use case for this integration and alternatives you've tried:**

Please add board definition to use with wizard

**Additional context**

OttoWinter, 2019-11-26:
For that device, just select the "generic" option as described in the setup wizard.

The board definition is not needed for this board - it has no pin aliases and has a standard WROOM-32 module.

As for the display, that is another thing - but for that create a separate feature request.

closed 2019-11-26 by OttoWinter

KamilLenica, 2019-11-26:
I will try to add support myself and contribute to the project