esphome/feature-requests #500: CCS811 Temperature and Humidity Compensation

ferazambuja, 2019-11-24:
**Describe the problem you have/What new integration you would like**

CCS811 uses gas readings to calculate eCO2 and TVOC. Changes in temperature and humidity affect the gas readings directly. To compensate for that CCS811 can use temperature and humidity values from an external sensor to calculates a more precise eCO2 and TVOC readings.

**Please describe your use case for this integration and alternatives you've tried:**

According to AMS:
[https://ams.com/documents/20143/36005/CCS811_DS000459_7-00.pdf/3cfdaea5-b602-fe28-1a14-18776b61a35a](url)
> Temperature and Humidity Compensation
If an external sensor is available this information can be written
to CCS811 so that they will be used to compensate gas readings
due to temperature and humidity changes. When ENV_DATA
has been written, the next eCO2 and eTVOC readings (in
ALG_RESULT_DATA) may not yet use the latest ENV_DATA. All
subsequent reading will use the ALG_RESULT_DATA. Refer to the
ENV_DATA (Environment Data) Register (0x05)

The ENV_DATA is already present on the esphome code for the component:
https://github.com/esphome/esphome/blob/be36eef939554478d5a48556827a5eed810de98d/esphome/components/ccs811/ccs811.cpp#L88

Here is the sparkfun implementation of the compensation.
https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/blob/master/examples/Example2_BME280Compensation/Example2_BME280Compensation.ino

**Additional context**

The sensor is also available as a combo BME280/CSS811 (https://www.sparkfun.com/products/14348) and HDC1080/CSS811 (Chinese). Both temperature sensors are supported by esphome.
However, such a combo will probably yield temperature values way higher due to the heating element on the CSS811. And the further investigation is needed, I could report here since I own the sparkfun combo. 

> The BME280 is unable to compensate for the heat produced by the CCS811 This can cause the displayed temperature to be up to 15° higher than the actual ambient temperature. To compensate for this take the difference between an initial temperature reading (with a cold sensor) and a final temperature reading (after letting the sensor warm up). Then subtract this value from `BMEtempC` to ensure the proper calibration is taking place.

https://learn.sparkfun.com/tutorials/ccs811bme280-qwiic-environmental-combo-breakout-hookup-guide

OttoWinter, 2019-11-24:
That is already supported. Please read the documentation page of the CCS811 carefully - see advanced options.

As for combination with other sensor heating up: You can compensate for that by using sensor filters.

closed 2019-11-24 by OttoWinter

ferazambuja, 2019-11-24:
Missed, sorry.