esphome/feature-requests #501: Non-integer font size

escoand, 2019-11-26:
**Describe the problem you have/What new integration you would like**
I try to use [Tetris Blocks Font](https://www.dafont.com/tetris-blocks.font) to show on a display but I struggle with the font size. When using 3 some blocks are 1px too large and with 2 some are 1px too small. Using a font size in between is currently not possible. The limitation to integer numbers seem not to exist in the Pillow library itself.

**Please describe your use case for this integration and alternatives you've tried:**
Is there a way to define the glyphs by hand? Maybe like this to get the `L` block:
```yaml
font:
  - id: tetris
    glyphs:
      L:
        - [1, 0]
        - [1, 0]
        - [1, 1]
```

**Additional context**

escoand, 2019-11-26:
Ok, it's already doing something like this:
```c
static const uint8_t uint8_t_3[] PROGMEM = {..., 0x80, 0x80, 0xC0, ...};
tetris = new display::Font({..., display::Glyph("J", uint8_t_3, 22, 0, 0, 2, 3), ...}, 2, 3);
```
Where the hex is in binary:
```
0x80 -> 1 0 0 0 0 0 0 0
0x80 -> 1 0 0 0 0 0 0 0
0xC0 -> 1 1 0 0 0 0 0 0
```
So, what is the best way to define this? Using a `*.h` file with `includes`?

OttoWinter, 2019-11-26:
> So, what is the best way to define this? Using a *.h file with includes?

You shouldn't. That's an API internal to ESPHome (and for example it expects some prerequisites, like the glyph array must be sorted).

Non-integer font size could be added, the validator in font/__init__.py just needs to be changed (and please also test the change then).

> Is there a way to define the glyphs by hand? 

1. let's keep this discussion on one thing only: non-integer font sizes
2. I believe you should be using `image` instead - I don't see why you should use font instead of images here.

closed 2019-12-10 by no-response[bot]