esphome/feature-requests #510: Allow same trigger with different parameters multiple times

dgtal1, 2019-11-29:
**Describe the problem you have/What new integration you would like**

Allow the same trigger e.g. `on_click` to occur multiple times for a component, but with different parameters.

**Please describe your use case for this integration and alternatives you've tried:**

I'm trying to have two different actions depending on how long a click lasts. 
E.g. 
1. normal, short click with default values -> turn on a switch (turn on a relay)
2. long click, min 5 seconds -> turn on another switch (restart)

Right now this is not possible, and I couldn't find a workaround.

**Additional context**

brandond, 2019-11-29:
You want `on_multi_click`: https://esphome.io/components/binary_sensor/index.html#on-multi-click

You can specify as many different timing combinations as you want.

brandond, 2019-11-29:
See an example here:
https://github.com/esphome/feature-requests/issues/472#issuecomment-559797780

closed 2019-11-30 by OttoWinter