esphome/feature-requests #511: Support ADXL345 sensor (digital accelerometer)

pfichtner, 2019-11-30:
**Describe the problem you have/What new integration you would like**

Support ADXL345 sensor (digital accelerometer)

**Please describe your use case for this integration and alternatives you've tried:**

Got it working using custom sensor
```
#include "esphome.h"
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_ADXL345_U.h>
  
class ADXL345Sensor : public PollingComponent {
 public:
  Adafruit_ADXL345_Unified accel = Adafruit_ADXL345_Unified(12345);
  Sensor *accel_x_sensor = new Sensor();
  Sensor *accel_y_sensor = new Sensor();
  Sensor *accel_z_sensor = new Sensor();

  ADXL345Sensor() : PollingComponent(1000) { }

  void setup() override {
    accel.begin();
    /* Set the range to whatever is appropriate for your project */
    // accel.setRange(ADXL345_RANGE_16_G);
    accel.setRange(ADXL345_RANGE_8_G);
    // accel.setRange(ADXL345_RANGE_4_G);
    // accel.setRange(ADXL345_RANGE_2_G);
    //
  }

  void update() override {
    sensors_event_t event;
    accel.getEvent(&event);

    accel_x_sensor->publish_state(event.acceleration.x);
    accel_y_sensor->publish_state(event.acceleration.y);
    accel_z_sensor->publish_state(event.acceleration.z);
  }

};

```

```
- platform: custom
lambda: |-
    auto adxl345 = new ADXL345Sensor();
    App.register_component(adxl345);
    return { adxl345->accel_x_sensor, adxl345->accel_y_sensor, adxl345->accel_z_sensor };
sensors:
- name: "Acceleration X"
- name: "Acceleration Y"
- name: "Acceleration Z"
```

**Additional context**

[Adafruit-ADXL345](https://platformio.org/lib/show/15/Adafruit-ADXL345-Unified)

twochannel, 2020-01-07:
The adxl345 has a cool feature where you can have it simply tell if you there is activity or not instead of watching the 3 axis.  Is it possible to have this support the motion/activity feature?

from the [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/ADXL345.pdf): "Several special sensing functions are provided. Activity and inactivity sensing detect the presence or lack of motion by comparing the acceleration on any axis with user-set thresholds. "

Additionally there is also an [adxl343 ](https://learn.adafruit.com/adxl343-breakout-learning-guide)that adafruit  states is basically identical. Can this also be supported?

cybermaus, 2022-11-12:
So I got this to work, but it needed some tweaking. Partly because I may misunderstood it, partly because Adafruit seemed to have rather rewritten parts of the library:

The custom source remains pretty much unchanged, just drop this in your ESPhome device folder
```haskell
#include "esphome.h"
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_ADXL345_U.h>
  
class ADXL345Sensor : public PollingComponent {
 public:
  Adafruit_ADXL345_Unified accel = Adafruit_ADXL345_Unified(12345);
  Sensor *accel_x_sensor = new Sensor();
  Sensor *accel_y_sensor = new Sensor();
  Sensor *accel_z_sensor = new Sensor();

  ADXL345Sensor() : PollingComponent(10000) { }

  void setup() override {
    accel.begin();
    /* Set the range to whatever is appropriate for your project */
    // accel.setRange(ADXL345_RANGE_16_G);
    // accel.setRange(ADXL345_RANGE_8_G);
    // accel.setRange(ADXL345_RANGE_4_G);
    accel.setRange(ADXL345_RANGE_2_G);
    //
  }

  void update() override {
    sensors_event_t event;
    accel.getEvent(&event);

    accel_x_sensor->publish_state(event.acceleration.x);
    accel_y_sensor->publish_state(event.acceleration.y);
    accel_z_sensor->publish_state(event.acceleration.z);
  }

};
```

But for the yaml, I had to include different libraries:
```haskell
esphome:
...
  includes:
    - ADXL345.h
  libraries:
    - "Wire"
    - "SPI"
    - "Adafruit BusIO"
    - "Adafruit Unified Sensor"
    - "Adafruit ADXL345"

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true

sensor:
...
  - platform: custom
    lambda: |-
      auto adxl345 = new ADXL345Sensor();
      App.register_component(adxl345);
      return { adxl345->accel_x_sensor, adxl345->accel_y_sensor, adxl345->accel_z_sensor };
    sensors:
      - name: "Acceleration X"
      - name: "Acceleration Y"
      - name: "Acceleration Z"

   

```

B48D81EFCC, 2023-10-15:
Hi,

is there a timeline to add this sensor to esphome "officially"?

bucovaina, 2023-10-21:
I would love this sensor to be included as well!

B48D81EFCC, 2023-10-21:
Hi again,

I found this code repository, which seems to have a bit more elaborated code of the snippet above:
https://codeberg.org/derflob/esphome-packages/src/branch/main/baresp/ADXL345.h

Unfortunately I have no experience with python nor C++.
I don't trust myself to be able to use this and contribute this to esphome. Anyone else?