esphome/feature-requests #513: Support for SD-Card as data sink

pfichtner, 2019-11-30:
**Describe the problem you have/What new integration you would like**

I'd like to create a standalone sensor (no WiFi available) that persists the sensor values read to a connected sd-card. 

**Please describe your use case for this integration and alternatives you've tried:**

There is no support in esphome for writing values to SD-cards so nothing tried yet. Is this a scenario you'd like to support? As far as I see esphome seems to support only mqtt for now. 

**Additional context**

Misiu, 2019-12-02:
@pfichtner try https://github.com/xoseperez/espurna or https://github.com/letscontrolit/ESPEasy.
I think ESPHome should stay as simple as possible and do one thing well - communicate with Home Assistant. Data logging should be done there

OttoWinter, 2019-12-04:
@Misiu I know my opinion on this has probably changed a bit over time. But I do think this is a valid use-case - having a constant connection to devices is not always possible (and does tend to require a lot of energy).

There's currently not great support for standalone devices, even though they could be great for quick data gathering in remote locations.

I'm not saying I will implement it - SD could be semi-hard to do depending on the filesystem APIs (haven't tried those yet). But if someone creates a good, clean implementation of SD card as a data sink for sensors for example, I'd be ok with merging it.

lennert1986, 2020-03-29:
Hi. I would like to have a SD card option. Looking forward for this solution one day ;-)

TheGroundZero, 2021-04-17:
`+1`'ing this, specifically for ESP32-CAMs with microSD card slots

kotyara75, 2021-10-03:
+1, some of my sensors have connection to home assistant just couple of times a day, it would be great if ESP-Home could cache events to sd card, then upload them to home assistant when connection is restored.

dbuggz, 2022-04-11:
I have a sensor in my 4WD that monitors the Auxiliary Battery. It currently saves data to an SD card and can display that on a WWW page when it is connected to my Home Network. Is there a way to integrate it into ESPhome  ?

nagyrobi, 2022-07-08:
Also see https://github.com/esphome/feature-requests/issues/1560

langfeld, 2022-09-20:
Is there any news on this? It would be great if you could save to an SD card instead of the internal memory. I just made a gas meter, which would have to save its value quite often.

adamlove86, 2023-08-18:
+1 woudl be great - especially for example, for esp32-cam where 2.5g wifi can get pretty congested at times - especially if there are multiple feeds which currently have to be monitored from server rather than on-board.

michaelsteinmann, 2023-08-29:
I would like to play MP3 files stored on an SD card through the media_player component of esphome.

ryanm101, 2023-10-30:
I have a use case for this too.

Gas Meter connected to esp. If power to esp is cut(not to whole house) and gas is in use then i have no way of knowing what the usage was in that period without manual reset of data. 
Battery backup + SD support would allow logging of data during that period and would enable the data to be logged on reconnect.

ptr727, 2023-11-17:
Following, I have a [SC-EN-I6-RO4](https://sensoper.com/shop/sc-en-i6-ro4/) DIN mount ESP32 that awaits microSD card support.

Larsuu, 2023-11-19:
Following. I just had an idea of turning JBM BMS with Esphome into Black Box for lithium ion battery. So it would need to store all the data of how it was used and then syncing it with home assistant when available. Ofcourse offline data would need to be stored meanwhile on SD card.  

Maybe eeprom memory could be used as saving details of the battery in case of emergency and then the previous records could be used along these. So maybe Flash memory and eeprom could do, if data logging is restricted to certain limits and only storing big changes? Depends on how much data there is flowing.

RrPt, 2024-03-14:
Would bei a nice Feature.
Any News?

pebblebed-tech, 2024-03-28:
Please see https://github.com/esphome/esphome/pull/6071 which provides external storage to a I2C eeprom which would address some of the use cases identified in feature request

Larsuu, 2024-04-04:
> Would bei a nice Feature. Any News?

Talkin about my idea? 

i did go more in-depth about it here.( https://github.com/syssi/esphome-jbd-bms/discussions/66 ) But, it's just an idea. Currently, I don't have any time for this idea. I could pull it into my project as an add-on if possible.

elproko, 2024-12-11:
Bumping the subject.
I think its shame no one from project owners discovered advantages for it.

I see this as ability to temporarily store sensor data on SD card and then upload it to HA as kind of "historical" with proper timestamps. I suspect this mode could have some data processing limitations in HA. (just because data is from the past) 

I found custom made component [sd_mmc_card](https://github.com/n-serrette/esphome_sd_card) written by [Nicolas Serrette](https://github.com/n-serrette) which supports writing and reading from SD card.
I've tested this component with esp-cam module and it seems it works with basic functionality. 

I would see this incorporated into ESPHome with use of native SPI component.

TCB13, 2025-02-10:
+1 on this.

paradisgaston, 2025-02-17:
I would like, this, to store sensors for a project using LoRa communication .