esphome/feature-requests #515: Support for DS3231 (I2C RTC)

49616e, 2019-12-01:
**Describe the problem you have/What new integration you would like**

I understand supporting this should not be a priority over other useful sensors, but having support for some kind of reliable I2C RTC chip would be nice in certain situations, especially if the ESPHome node is placed at a location where the wifi signal strength is poor.
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

lsaoye, 2020-01-26:
Hope to get support

rradar, 2020-01-27:
This could go very nice along with a ntp server component https://github.com/esphome/feature-requests/issues/296 :tada:

e-budianto, 2020-03-06:
hopefully there will be support for ds1307 too

a-x-, 2020-05-09:
Sometimes electricity breaks in our location and then restores, but internet connection restoring in a long time after usually, so our ESP modules cannot understand the time for light controlling logic

Denyerec, 2020-10-06:
Just been looking for ways to have backup programming for my boiler and an RTC was a requirement - would love to see it!

wisesokol, 2020-11-09:
Me too. If there is no wi fi, I would like to show the correct time on the display.

badbadc0ffee, 2021-01-03:
> hopefully there will be support for ds1307 too

I have finally started working on that again:
https://github.com/esphome/esphome/pull/1441

badbadc0ffee, 2021-01-04:
The DS3231 register set is so similar to the one of the DS1307, that the DS1307 component (esphome/esphome#1441) should already work with a DS3231. I could not test this, however, because I don't have such a device at hand; and additional features such as temperature readings and alarms are obviously not supported.

I could definitely also provide a dedicated driver component for the DS3231 if I had one to test with. :-)
https://www.amazon.de/hz/wishlist/ls/IKJQ9RPIUL6J

rradar, 2021-01-11:
https://github.com/esphome/esphome/pull/1441 is merged in [1.16.0b2](https://github.com/esphome/esphome/releases/tag/v1.16.0b2).

@badbadc0ffee I need to find my DS3231 somewhere in the drawer so I can test it with your component! Thank's already for your work :+1:

NarinLab, 2021-02-08:
Any update on DS3231 support for ESP32? Or if someone can show us how ti write our own code porting from conventional Arduino IDE based approach to get DS3231 integrated into ESPHOME.

jbrepogmailcom, 2021-02-17:
I would like if I can wake ESP by external clock to do its job and then turn it off completely. It is not conceptually correct to have ESP sleeping and all supporting circuitry powered up if no job is needed. Everything should be powered down until its time comes.

ckuethe, 2021-02-17:
@jbrepogmailcom I think you misunderstand. It is perfectly reasonable to leave the rest of the external support circuitry powered up when calling a routine that is documented to put the processor to sleep. Switching off power to the entire board is beyond the scope of the the deep sleep routine. You could probably build a basic switch from a couple of transistors, or use a dedicated power management IC (eg. AXP192)

jbrepogmailcom, 2021-02-17:
Hello, we are probably talking about different things. I meant to switch off both circuitry and ESP32 by DS3231 during the deep sleep, not before. And after wakeup, to reprogram DS3231 for nex sleep. However, I learned in the meanwhile that DS3231 takes about 100µA during sleep, while ESP32 with RTC on takes 10µA in deep sleep and 5µA in hibernation mode. So it is better to use ESP32 itself to manage sleep time and turn on/off supporting electronics.

zenzay, 2021-03-31:
A bit late to the discussion, but the DS1307 component in ESPHome seems to work just fine with the DS3231 module. I have it attached to a NodeMCU 32s board and it's _ticking_ along nicely.

WeekendWarrior1, 2021-09-15:
> Edit: I'm getting two devices at the i2c scan. I'm, ofc, using 0x68. I have no idea what the other is tbh.

The other is likely 32K of eeprom (AT24C32X), which I know can have an address of 0x57 and appears on modules like this:

[image]

Writing to and reading from it is not currently supported by esphome.

zenzay, 2021-09-15:
That seems very likely, as that's the exact module I have :-)

I currently have no use for the eeprom on the module, so it's not really something I would wish esphome to focus on anyhow. 

Thanks for the information!

kevindorff, 2021-10-13:
Just to throw my two cents in. For my ESPHome sensors, I'm hoping to replicate the work found https://getpocket.com/read/3454710576 to greatly prolong the battery life on my sensors. It seems like this "sleep" method allows for zero current to the DS321 (via VCC) or ESP32 and very, very low current for the P-Channel Mosfet switch (< 1mA). The DS3231 runs from the coin cell while waiting for the next alarm. When that alarm happens, the DS3231 switches the mosfet, the ESP32 circuit is powered on. The ESP32 can then do it's business, set a next alarm, and when it clears the current DS3231 alarm the ESP32 circuit is shutdown (completely), along with the VCC input on the DS3231 (so it is only running from the coin-cell until it triggers the next alarm, turning the ESP32 circuit back on).

It seems like several others have more or less the same goal. Has anyone made any progress?

I first will probably just do it as an Arduino or MicroPython app and publish to Home Assistant via MQTT (via node-red probably). I'd love to then transfer what I've learned to be able to do this with ESPHome, but we'll see how far I get on that front. It's currently not clear to me how I'd **want** to implement this in ESPHome. Maybe as an alternative to deep sleep or via lambda scripts?

nagyrobi, 2022-07-01:
https://esphome.io/components/time.html?#ds1307-time-source

kevindorff, 2022-07-01:
> https://esphome.io/components/time.html?#ds1307-time-source

That's an interesting device for reading/writing time. The ds3231 could enable doing a lot more. See my description above. 

My goal is to be able to use the ds3231 to be able to be able to **power down** circuit including the esp32 for stretches of time and it can trigger turning the circuit back on to do work - and the cycle repeats.

ptr727, 2023-11-17:
Following, I have a [SC-EN-I6-RO4](https://sensoper.com/shop/sc-en-i6-ro4/) DIN mount ESP32 that awaits DS3231 support.

elspru, 2024-12-30:
okay guys got it figured out with esp32-c3 super mini anyhow. see my esphome.yaml files attached one for setting the time, the other for reading the time. 
[ds3231-esphome-scripts.zip](https://github.com/user-attachments/files/18277930/ds3231-esphome-scripts.zip)

htaf@mriczo:~/liberit/kyut/ds3231-rtc$ cat ds3231-set-time.yaml 
esphome:
  name: ds3231-set-time

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

i2c:
  sda: GPIO20
  scl: GPIO21
  scan: true

logger:

wifi:
  output_power: 8.5dB
  networks:
    - ssid: "yourWifiNetwork"
      password: "yourPassword"

time:
  - platform: sntp
    id: sntp_time
    timezone: America/Toronto

custom_component:
  - lambda: |-
      class DS3231Component : public PollingComponent {
       public:
        DS3231Component(esphome::time::RealTimeClock *rtc) : PollingComponent(10000), rtc_(rtc) {}

        void setup() override {
          if (!Wire.requestFrom(0x68, 1)) {
            ESP_LOGE("DS3231", "RTC not found at I2C address 0x68");
            return;
          }
          ESP_LOGI("DS3231", "RTC found at I2C address 0x68");

          auto now = this->rtc_->now();
          if (now.is_valid()) {
            set_time(now.year, now.month, now.day_of_month, now.hour, now.minute, now.second, now.day_of_week);
            ESP_LOGI("DS3231", "RTC time set to %04d-%02d-%02d %02d:%02d:%02d",
                     now.year, now.month, now.day_of_month, now.hour, now.minute, now.second);
          } else {
            ESP_LOGE("DS3231", "SNTP time is not valid; cannot set RTC");
          }
        }

        void update() override {
          auto time = get_time();
          if (time.year > 0) {
            ESP_LOGI("DS3231", "Current RTC Time: %04d-%02d-%02d %02d:%02d:%02d",
                     time.year, time.month, time.day, time.hour, time.minute, time.second);
          } else {
            ESP_LOGE("DS3231", "Failed to read time from RTC");
          }
        }

        void set_time(int year, int month, int day, int hour, int minute, int second, int day_of_week) {
          Wire.beginTransmission(0x68);
          Wire.write(0);  // Start at register 0
          Wire.write(dec_to_bcd(second));        // Seconds
          Wire.write(dec_to_bcd(minute));        // Minutes
          Wire.write(dec_to_bcd(hour));          // Hours
          Wire.write(dec_to_bcd(day_of_week));   // Day of the week
          Wire.write(dec_to_bcd(day));           // Day of the month
          Wire.write(dec_to_bcd(month));         // Month
          Wire.write(dec_to_bcd(year - 2000));   // Year
          Wire.endTransmission();
        }

        struct Time {
          int year;
          int month;
          int day;
          int hour;
          int minute;
          int second;
          int day_of_week;
        };

        Time get_time() {
          Wire.beginTransmission(0x68);
          Wire.write(0);  // Start at register 0
          Wire.endTransmission();

          if (Wire.requestFrom(0x68, 7) != 7) {
            ESP_LOGE("DS3231", "Failed to read time registers");
            return Time{0, 0, 0, 0, 0, 0, 0};
          }

          uint8_t second = bcd_to_dec(Wire.read());
          uint8_t minute = bcd_to_dec(Wire.read());
          uint8_t hour = bcd_to_dec(Wire.read());
          uint8_t day_of_week = bcd_to_dec(Wire.read());
          uint8_t day = bcd_to_dec(Wire.read());
          uint8_t month = bcd_to_dec(Wire.read());
          uint16_t year = bcd_to_dec(Wire.read()) + 2000;

          return Time{year, month, day, hour, minute, second, day_of_week};
        }

       private:
        esphome::time::RealTimeClock *rtc_;

        uint8_t dec_to_bcd(int val) {
          return ((val / 10 * 16) + (val % 10));
        }

        int bcd_to_dec(uint8_t val) {
          return ((val / 16 * 10) + (val % 16));
        }
      };

      auto my_rtc = new DS3231Component(id(sntp_time));
      App.register_component(my_rtc);
      return {};

htaf@mriczo:~/liberit/kyut/ds3231-rtc$ 

htaf@mriczo:~/liberit/kyut/ds3231-rtc$ cat ds3231-read-time.yaml 
esphome:
  name: ds3231-read-time
  platform: ESP32
  board: esp32-c3-devkitm-1

i2c:
  sda: GPIO20  # Adjust to your setup
  scl: GPIO21
  scan: true

logger:

# Use the DS1307 platform to interact with the DS3231 RTC
time:
  - platform: ds1307
    id: rtc_time
    timezone: America/Toronto
    on_time_sync:
      then:
        - logger.log: "Time synchronized with RTC."
        - lambda: |-
            auto now = id(rtc_time).now();
            ESP_LOGI("RTC", "Current time: %04d-%02d-%02d %02d:%02d:%02d",
                     now.year, now.month, now.day_of_month, now.hour, now.minute, now.second);

interval:
  - interval: 5s
    then:
      - lambda: |-
          auto now = id(rtc_time).now();
          ESP_LOGI("Time", "Current time: %04d-%02d-%02d %02d:%02d:%02d",
                   now.year, now.month, now.day_of_month, now.hour, now.minute, now.second);

htaf@mriczo:~/liberit/kyut/ds3231-rtc$

elspru, 2024-12-30:
oh and hey you guys wanted the alarm via SQW pin, here is how to set it up, for every minute on the first second, check the set_alarm() code if you need a different interval:
```
esphome:
  name: ds3231-alarm-test
  platform: ESP32
  board: esp32-c3-devkitm-1

i2c:
  sda: GPIO20  # Adjust to your wiring
  scl: GPIO21
  scan: true

logger:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO9  # Pin connected to SQW/INT from DS3231
      mode: INPUT_PULLUP
    name: "DS3231 Alarm Trigger"
    filters:
      - delayed_off: 10ms  # Debounce to avoid false positives
    on_press:
      - logger.log: "Alarm Detected!"  # Log when interrupt occurs

custom_component:
  - lambda: |-
      class DS3231Alarm : public Component {
       public:
        void setup() override {
          if (!Wire.requestFrom(0x68, 1)) {
            ESP_LOGE("DS3231", "RTC not found at I2C address 0x68");
            return;
          }
          ESP_LOGI("DS3231", "RTC found at I2C address 0x68");

          // Set up the alarm to trigger every 6 seconds
          set_alarm();
        }

        void set_alarm() {
          Wire.beginTransmission(0x68);
          Wire.write(0x07);        // Start at Alarm1 seconds register
          Wire.write(0x01);        // A1M1=0: Match on first second
          Wire.write(0x80);        // A1M2=1: Match any minute
          Wire.write(0x80);        // A1M3=1: Match any hour
          Wire.write(0x80);        // A1M4=1: Match any day
          Wire.endTransmission();

          // Enable Alarm1 interrupt and disable square wave
          Wire.beginTransmission(0x68);
          Wire.write(0x0E);  // Control register
          Wire.write(0x05);  // INTCN=1 (enable interrupt), A1IE=1 (enable Alarm1)
          Wire.endTransmission();

          ESP_LOGI("DS3231", "Alarm set to trigger every 6 seconds.");
        }

        void loop() override {
          // Check if Alarm1 triggered by reading the status register
          Wire.beginTransmission(0x68);
          Wire.write(0x0F);  // Status register
          Wire.endTransmission();
          Wire.requestFrom(0x68, 1);
          uint8_t status = Wire.read();

          if (status & 0x01) {  // Alarm1 flag (A1F) is set
            ESP_LOGI("DS3231", "Alarm Triggered!");

            // Clear the Alarm1 flag
            Wire.beginTransmission(0x68);
            Wire.write(0x0F);  // Status register
            Wire.write(status & ~0x01);  // Clear A1F
            Wire.endTransmission();
          }
        }
      };

      auto my_alarm = new DS3231Alarm();
      App.register_component(my_alarm);
      return {};
```

dj-fiorex, 2025-03-07:
@elspru thanks for the example, does this work with an esp32-s3 and esp-idf framework?
i have this config:
```
substitutions:
  device_name: "v"
  friendly_name: "V"
  ip_address: 192.168.188.190
  device_description: "Project v. - IP: ${ip_address}"

esphome:
  name: ${device_name}
  comment: ${device_description}
  friendly_name: ${friendly_name}
  min_version: 2024.11.0
  name_add_mac_suffix: false
  libraries:
    # a library bundled with Arduino
    - Wire

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: !secret api_encryption_key

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Optional manual IP
  manual_ip:
    static_ip: ${ip_address}
    gateway: 192.168.188.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}-FH
    password: !secret ap_password

i2c:
  sda: GPIO48
  scl: GPIO47
  scan: true
  id: bus_a

sensor:

  # BME280
  - platform: bme280_i2c
    temperature:
      name: "BME280 Temperature"
      id: bme280_temperature
    pressure:
      name: "BME280 Pressure"
      id: bme280_pressure
    humidity:
      name: "BME280 Relative Humidity"
[... 100 lines cut ...]
and when i compile the program i get this error:
```
Library Manager: Installing Wire
Warning! Could not find the package with 'Wire' requirements for your system 'linux_x86_64'
Library Manager: Installing Wire
Warning! Could not find the package with 'Wire' requirements for your system 'linux_x86_64'
Dependency Graph
........
........
Compiling .pioenvs/vaso-smart/console/linenoise/linenoise.c.o
/root/config/vaso-smart.yaml: In member function 'virtual void setup()::<lambda()>::DS3231Alarm::setup()':
/root/config/vaso-smart.yaml:116:16: error: 'Wire' was not declared in this scope
  116 |           if (!Wire.requestFrom(0x68, 1)) {
      |                ^~~~
/root/config/vaso-smart.yaml: In member function 'void setup()::<lambda()>::DS3231Alarm::set_alarm()':
/root/config/vaso-smart.yaml:127:11: error: 'Wire' was not declared in this scope
  127 |           Wire.beginTransmission(0x68);
      |           ^~~~
/root/config/vaso-smart.yaml: In member function 'virtual void setup()::<lambda()>::DS3231Alarm::loop()':
/root/config/vaso-smart.yaml:147:11: error: 'Wire' was not declared in this scope
  147 |           Wire.beginTransmission(0x68);
      |           ^~~~
Compiling .pioenvs/vaso-smart/console/argtable3/arg_cmd.c.o
```
Thanks!

miloit, 2025-03-29:
Hell i have the config

`globals:
  - id: last_on_relayone_timestamp
    type: std::string
    restore_value: true
    initial_value: '"Never ON"'

  - id: last_off_relayone_timestamp
    type: std::string
    restore_value: true
    initial_value: '"Never OFF"'
  
  - id: last_on_relaytwo_timestamp
    type: std::string
    restore_value: true
    initial_value: '"Never ON"'

  - id: last_off_relaytwo_timestamp
    type: std::string
    restore_value: true
    initial_value: '"Never OFF"'
    
  - id: last_on_relaythree_timestamp
    type: std::string
    restore_value: true
    initial_value: '"Never ON"'

  - id: last_off_relaythree_timestamp
    type: std::string
    restore_value: true
    initial_value: '"Never OFF"'
    
  - id: last_on_relayfour_timestamp
    type: std::string
    restore_value: true
    initial_value: '"Never ON"'

  - id: last_off_relayfour_timestamp
    type: std::string
    restore_value: true
    initial_value: '"Never OFF"'
    
    
substitutions:
  # Default name
  name: "wateringdevice"
  # Default friendly name
  friendly_name: "wateringdevice"
  # Allows ESP device to be automatically linked to an 'Area' in Home Assistant. Typically used for areas such as 'Lounge Room', 'Kitchen' etc
  room: ""
  # Description as appears in ESPHome & top of webserver page
  device_description: "wateringdevice"
  # Project Name
  project_name: "wateringdevice"
  # Projection version denotes the release version of the yaml file, allowing checking of deployed vs latest version
  project_version: "v1.0.3"
   # Restore the relay (GPO switch) upon reboot to state:
  relay1_restore_mode: RESTORE_DEFAULT_OFF
  relay2_restore_mode: RESTORE_DEFAULT_OFF
  relay3_restore_mode: RESTORE_DEFAULT_OFF
  relay4_restore_mode: RESTORE_DEFAULT_OFF
  # Define a domain for this device to use. i.e. iot.home.lan (so device will appear as athom-smart-plug-v2.iot.home.lan in DNS/DHCP logs)
  dns_domain: ".local"
  # Set timezone of the smart plug. Useful if the plug is in a location different to the HA server. Can be entered in unix Country/Area format (i.e. "Australia/Sydney")
  timezone: "Europe/Berlin"
  # Set the duration between the sntp service polling ntp.org servers for an update
  sntp_update_interval: 6h
  # Network time servers for your region, enter from lowest to highest priority. To use local servers update as per zones or countries at: https://www.ntppool.org/zone/@
  sntp_server_1: "0.pool.ntp.org"
[... 495 lines cut ...]
    name: "Relais2 Zeit (%)"
    id: relais_two_zeit_prozent
    min_value: 0
    max_value: 100
    step: 1
    optimistic: true
    mode: slider

i2c:
  sda: GPIO21  # Passe diese Pins an dein Setup an
  scl: GPIO22
  scan: true   # Zeigt I²C-Geräte im Log`

but I get this error

`watering.yaml:501:30: error: 'DS3231Component' does not name a type
watering.yaml:501:45: error: expected '>' before '*' token
watering.yaml:501:45: error: expected '(' before '*' token
watering.yaml:501:46: error: expected primary-expression before '>' token
watering.yaml:501:72: error: 'class esphome::Component' has no member named 'get_time'; did you mean 'set_timeout'?
watering.yaml:501:82: error: expected ')' before ';' token
*** [.pioenvs/wateringdevice/src/main.cpp.o] Error 1
`

how do i call the function correctly?

zubrick, 2025-05-28:
As custom components are not a possibility anymore and I needed a to have my DS3231 fully work with for porting my led matrix project to esphome, I converted the rtc sync and get time to an external component, you just have to add this to your yaml:
```yaml
external_components:
  - source: github://zubrick/zmatrix-esphome@main
    components: [ ds3231 ]
    refresh: 10min

ds3231:
  id: ds3231_time
  time_id: sntp_time
  update_interval: 240min

time:
  - platform: sntp
    id: sntp_time
    timezone: "Europe/Zurich"
```

the update_interval parameter lets you decide how often you want to try syncing the rtc to the ntp (or any other time source you may use)

you can then call ```id(ds3231_id).get_time()``` in your lambda to get the rtc time returned as the Time structure

Sorry, I didn't implement the alarm part, I don't have any use for that