esphome/feature-requests #517: xiaomi bluetooth kettle

ppanagiotis, 2019-12-01:
**Describe the problem you have/What new integration you would like**

Use ble device to connect with xiaomi kettle

**Please describe your use case for this integration and alternatives you've tried:**

A nice try is described here
https://github.com/aprosvetova/xiaomi-kettle

**Additional context**

n4ts, 2019-12-17:
This would be a nice feature-requests :)
If we would like to pull only the temperature then we can use the solution for Xiaomi LYWSDCGQ or Xiaomi CGG1 which works the same (they take the temperature based on BLE data).

Simple code is below:
https://github.com/able99/minik-ha-components/blob/master/kettle_mi/custom_components/sensor/kettle_mi.py

jascdk, 2020-02-02:
Great - I have thought about the same:) It would be a cool integration:)

Titanius, 2020-04-09:
**Additional context**

andriej, 2020-05-20:
No further news?

WildeRNS, 2020-12-12:
There is working custom [component](https://github.com/custom-components/ble_monitor)  that uses passive ble, maybe now it can be added to esphome?

makefu, 2021-01-18:
I'd love to see this device being supported!

n4ts, 2021-01-23:
For me, even getting temperature in the passive mode would be sufficient ;) maybe we can do this by getting advertise data?

n4ts, 2021-02-01:
If someone needs to get only the temperature (in passive mode), below is my ready code to use:
`esp32_ble_tracker:
  scan_parameters:
    duration: 10s
    active: false
  on_ble_service_data_advertise:
    - mac_address: YOUR_MAC_HERE
      service_uuid: FE95
      then:
        - lambda: |-
            id(kettle_data).publish_state(float(x[16]));

sensor:
  - platform: template
    id: kettle_data
    name: "Kettle Temp"
    unit_of_measurement: '°C'
    accuracy_decimals: 0
    icon: "mdi:thermometer"`

Maybe it could be implemented in esphome as ready solution?

Hilfarnal, 2021-02-01:
@n4ts Could you please upload all of your code? I'm new user and I'm geting syntax errors I can't fix.

n4ts, 2021-02-01:
@Hilfarnal Below is the code and a small feature;)
```YAML
esp32_ble_tracker:
  scan_parameters:
    duration: 10s
    active: false
  on_ble_service_data_advertise:
    - mac_address: YOUR_MAC_HERE
      service_uuid: FE95
      then:
        - lambda: |-
            id(kettle_temp).publish_state(float(x[16]));
            if(x[15] == 01) {
              id(kettle_mode).publish_state(true);
            } else {
              id(kettle_mode).publish_state(false);
            }

sensor:
  - platform: template
    id: kettle_temp
    name: "Kettle Temp"
    unit_of_measurement: '°C'
    accuracy_decimals: 0
    icon: "mdi:thermometer"

binary_sensor:
  - platform: template
    id: kettle_mode
    name: "Kettle Mode"
```

I tried to decode the rest of the advertise data, but apart from the temperature, mode and MAC address, I wasn't able to decode the rest of the values. Even after using the information contained here https://github.com/aprosvetova/xiaomi-kettle
Please write if you have any fixes for my code.

jascdk, 2021-02-01:
> @Hilfarnal Below is the code and a small feature;)
> 
> ```yaml
> esp32_ble_tracker:
>   scan_parameters:
>     duration: 10s
>     active: false
>   on_ble_service_data_advertise:
>     - mac_address: YOUR_MAC_HERE
>       service_uuid: FE95
>       then:
>         - lambda: |-
>             id(kettle_temp).publish_state(float(x[16]));
>             if(x[15] == 01) {
>               id(kettle_mode).publish_state(true);
>             } else {
>               id(kettle_mode).publish_state(false);
>             }
> 
> sensor:
>   - platform: template
>     id: kettle_temp
>     name: "Kettle Temp"
>     unit_of_measurement: '°C'
>     accuracy_decimals: 0
>     icon: "mdi:thermometer"
> 
> binary_sensor:
>   - platform: template
>     id: kettle_mode
>     name: "Kettle Mode"
> ```
> 
> I tried to decode the rest of the advertise data, but apart from the temperature, mode and MAC address, I wasn't able to decode the rest of the values. Even after using the information contained here https://github.com/aprosvetova/xiaomi-kettle
> Please write if you have any fixes for my code.
> 
> Example and description of received data below:
> 95fe712013010fzzzzzzzzzzzz09051002yyxx
> Where:
> xx - temp
> yy - mode
> zz - reverse mac address
> 95fe is service UUID

Wow this is great - works perfect:) Looking forward to se if anyone comes up with decoding of the rest of the values:)

By the way - how did you sniff the data-strings? Also with ESPHome ble tracker?

Hilfarnal, 2021-02-02:
@n4ts @jascdk After a little bit of studying esphome web and learning proper syntax i got it working yesterday. Thanks for your contribution!

jascdk, 2021-02-03:
@n4ts Hi :) A little bit off topic, but I as said I got your code to get working - great:) I have an Xiaomi BLE toothbrush (Oclean x), which could be fun have reporting some stuff to HA. When I search for it with an BLE scanner on my Iphone I get the following.
[image]
[image]

Would it be possible to get the battery information with Lambda? And how should I do it - any help is appreciated :)

koenvervloesem, 2021-02-03:
@jascdk Does your toothbrush also advertise its battery status when you're not connected? If it does, this should do the trick (I'm using this for another device):

```
esp32_ble_tracker:
  on_ble_service_data_advertise:
    - mac_address: AB:CD:EF:01:23:45
      service_uuid: 180F
      then:
        - lambda: 'id(battery_sensor).publish_state(x[0]);'

sensor:
  - platform: template
    name: "Battery Sensor"
    id: battery_sensor
    accuracy_decimals: 0
    unit_of_measurement: '%'
    icon: 'mdi:battery'
```

jascdk, 2021-02-03:
> @jascdk Does your toothbrush also advertise its battery status when you're not connected? If it does, this should do the trick (I'm using this for another device):
>
> ```
> esp32_ble_tracker: [...]

Thanks :) Well - the app needs to connect to the brush to read the value. But when the brush is just idling it can also retrieve the information. But I did not get the code to get its value :( Any suggestion?

koenvervloesem, 2021-02-03:
The battery service according to the Bluetooth specs is 0x180F. So if your app doesn't show the service 180F when disconnected from the toothbrush, you can't use the code above. ESPHome can't connect yet to BLE devices, but if you want, you can try PR https://github.com/esphome/esphome/pull/1177 which implements this.

jascdk, 2021-02-04:
> The battery service according to the Bluetooth specs is 0x180F. So if your app doesn't show the service 180F when disconnected from the toothbrush, you can't use the code above. ESPHome can't connect yet to BLE devices, but if you want, you can try PR [esphome/esphome#1177](https://github.com/esphome/esphome/pull/1177) which implements this.

Nice thanks:) I think the problem is like what you telling. I have disconnected both the scanner, the original app and nothing happens, so it might be the issue. 

Im not so good at all this github / coding thing. How do i implement the PR in my existing Esphome thorugh home assistant? :)

koenvervloesem, 2021-02-04:
I don't think you can, but I don't use the ESPHome addon in Home Assistant. I'm using ESPHome standalone, installed with `pip`. You should be able to install it this way with something like `pip3 install git+https://github.com/buxtronix/esphome.git@ble-client` (not tested).

xrust83, 2021-07-16:
> sensor:
>   - platform: template
>     id: kettle_data
>     name: "Kettle Temp"
>     unit_of_measurement: '°C'
>     accuracy_decimals: 0
>     icon: "mdi:thermometer"
> ```
> 
> Maybe it could be implemented in esphome as ready solution?

The teapot sends the temperature too often, which spoils the log of my Homebridge, so I added a filter with a value of 1.0 celsius

`sensor:
  - platform: template
    id: kettle_temp
    name: "Kettle Temp"
    unit_of_measurement: 'C'
    accuracy_decimals: 0
    icon: "mdi:thermometer"
    filters:
      - delta: 1.0`

Now MQTT messages are sent only if the temperature changes by 1 degree Celsius

dentra, 2022-05-06:
I have already working solution for kittle component in my custom components xiaomi set https://github.com/dentra/esphome-components/tree/master/components/miot_kettle

pattyland, 2022-07-15:
> Hi, i want to have switch to enable boil maybe someone done this work?

It doesn't work that way and hopefully it will stay that way!

thathomieruffian, 2022-12-15:
> @Hilfarnal Below is the code and a small feature;)
>
> ```yaml
> esp32_ble_tracker:
>   scan_parameters:
>     duration: 10s
>     active: false
>   on_ble_service_data_advertise:
>     - mac_address: [...]

> @Hilfarnal Below is the code and a small feature;)
>
> ```yaml
> esp32_ble_tracker:
>   scan_parameters:
>     duration: 10s
>     active: false
>   on_ble_service_data_advertise:
>     - mac_address: [...]

managed to "decode" the states and output them to a text sensor:

```
# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:
  scan_parameters:
    duration: 10s
    active: True
  on_ble_service_data_advertise:
    - mac_address: "XX:XX:XX:XX:XX:XX"
      service_uuid: FE95
      then:
        - lambda: |-
            id(kettle_temp).publish_state(float(x[16]));
            if(x[15] == 01) {
              id(kettle_action).publish_state("heating");
            } else if(x[15] == 02) {
              id(kettle_action).publish_state("cooling");
            } else if(x[15] == 03) {
              id(kettle_action).publish_state("keep warm");
            } else if(x[15] == 00) {
              id(kettle_action).publish_state("idle");
            }

text_sensor: 
  - platform: template
    id: kettle_action
    name: "Kettle action" 
    update_interval: 5s
```

jacoqq, 2024-07-17:
Hi, 
Trough all intergations I've try that one is given me hope. There is something happend after I use this code. 

`[12:00:41][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:42][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:44][D][esp32_ble_tracker:270]: Starting scan...
[12:00:44][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:45][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:46][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:47][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:48][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:48][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:48][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:49][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:50][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:50][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:51][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:52][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:52][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:53][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:53][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:54][D][esp32_ble_tracker:270]: Starting scan...
[12:00:54][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:54][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:55][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy
[12:00:56][D][sensor:094]: 'Kettle Temp': Sending state 173.00000 °C with 0 decimals of accuracy`

But 173 deg is not correct value. Besides when water is boiling nothing in my log is change - still 173 deg. 
I'm preety sure I have connection, because when I grab my kette log stops and when I put it back data is readin again.
Can someone please help me and give me an advice what I need to do ?