esphome/feature-requests #518: [ESP32] BLE iTAG buttons

Underknowledge, 2019-12-01:
**Describe the problem you have/What new integration you would like**

For around 1€/$ you can get a dirt cheap Bluetooth LE device.
https://www.aliexpress.com/item/32837337050.html?spm=a2g0s.9042311.0.0.7d914c4d7Wd8wC
That thing got also an button... an ESP32 got Bluetooth. *Sigh* would be awesome to ring that thing trough an automation or trigger something like an scene.

**Please describe your use case for this integration and alternatives you've tried:**

Other projects already 

**Additional context**

1802 -> Immediate Alert 
1803 -> Link Loss
ffe0 -> Button
ffe1 -> Click
2a06 -> Alert Level 
0x00 -> no alert
0x01 -> mild alert (continuous)
0x02 -> high alert (beeping)

Sadly, the devices this project was tested on had no characteristics for battery level.

Misiu, 2019-12-02:
@Underknowledge I really like the idea but from my knowledge, those devices turn off automatically after 15 minutes when they are not connected (paired) via Bluetooth - https://github.com/1technophile/OpenMQTTGateway/issues/273#issuecomment-558801044
I ordered couple of those devices and will test them with ESP32.

I've ordered two of those: https://www.aliexpress.com/item/32866018860.html (because they don't have disconnect problem, or shouldn't have) and I'm considering ordering this: https://www.aliexpress.com/item/32907643142.html

Maybe pieces of https://github.com/1technophile/OpenMQTTGateway could be used?

Do you have those devices at home? Can you check the disconnect problem I mentioned?

Misiu, 2019-12-17:
@Underknowledge did you test those trackers with other projects? I'm still waiting for my order.
I'm curious about the disconnect problem.

Underknowledge, 2019-12-17:
@Misiu Christmas time is horror time. I only confirmed that the itag really goes in standby mode after a amount of time - this could be related to my almost empty battery. Also ordered the other ones. Waiting too. Got for testing a couple of tile trackers around. Next opening 28th DEC

sdesalve, 2019-12-22:
> @Underknowledge did you test those trackers with other projects? I'm still waiting for my order.
> I'm curious about the disconnect problem.

Cheapest iTag will turn off if not connected to phones.

I use Nut Mini for BtLE tracking with ESPHome and HomeAssistant with success

I'm searching a simple way to make my Nut ringing or read button pressure on other BtLE tags...

adrianmihalko, 2020-03-27:
I own iTag and Nut as well, let me know if I can help you.

Frucoboti, 2020-05-24:
Hello,
I am playing with NUT and Esphome, but to mix the esphome 'binary.sensor' device with the HA 'device_tracker' is a disaster. The 'device_tracker' is associated to a HA 'person' ID.
How do you do this?

This is the code at Esphome

_esp32_ble_tracker:
   scan_parameters:
     interval: 512ms
     window: 320ms
     duration: 300s
     active: true

binary_sensor:
  - platform: ble_presence
    device_class: presence
    mac_address: D6:19:04:59:11:D6
    name: "NUT_Ma"
    id: tracker_ma
    on_press:
      then:
      - homeassistant.service:
          service: device_tracker.see
          data:
            dev_id: tracker_ma
            location_name: 'home'
    on_release:
      then:
      - homeassistant.service:
          service: device_tracker.see
          data:
            dev_id: tracker_ma
            location_name: 'not_home'_           

But as I said, it not change the state.

Underknowledge, 2020-07-23:
Praise buxtronix
https://github.com/esphome/esphome/pull/1177

closed 2021-05-02 by jesserockz