esphome/feature-requests #519: DHT12 one wire support

ryrzy, 2019-12-05:
**Describe the problem you have/What new integration you would like**

please consider adding support for DHT12 one wire connection

**Please describe your use case for this integration and alternatives you've tried:**

i use dht12 sensors but in few devices i have connected them with one wire method, esphome dosnt support this. i use for now this sensor as dht11 and it work, but resolution is 1 °C and dht12 should have 0.1 °C resolution :) if it is possible please add one wire support for this sensor :)

**Additional context**

https://www.hackster.io/xreef/dht12-lib-for-humidity-and-temp-easy-lib-2-wire-for-all-e3b6ad

OttoWinter, 2019-12-07:
See https://github.com/esphome/esphome/pull/861

I think this is already supported in dev.

ryrzy, 2019-12-07:
how can i test this?

ryrzy, 2019-12-08:
ok, i have used dev branch and it looks like resolution is 0.1°C :) but still i must use DHT11 in config for my DHT12 :) it works, but it should be added to DHT12 to :P 
thanks for info