esphome/feature-requests #520: Opentherm support

rdehuyss, 2019-12-05:
**Describe the problem you have/What new integration you would like**

It would be great if ESPHome would support opentherm and directly integrate into Home Assistant

**Please describe your use case for this integration and alternatives you've tried:**

Currently I use the generic thermostat is HomeAssistant. My gas usage would probably diminish of I could use Opentherm modulation.

**Additional context**

brandond, 2019-12-07:
Do you have any reference documentation or a chipset that supports it? It looks like the signalling is done using voltage in one direction, and current in the other. Current sensing and control would require additional external circuitry.

rdehuyss, 2019-12-07:
My ideal situation would be that ESPHome could drive my central heating (including modulation) using opentherm.

There are some hardware options already available (http://ihormelnyk.com/opentherm_thermostat) including code for Arduino.

If we combine this with a smart thermostat in HomeAssistant (presence detection, weather forecast, input from various other sensors), I think we have a state of the art system.

There is also the opentherm gateway which can work using a serial connection over an ESP8266, could be another implemention of the same feature.

lwqcz, 2019-12-26:
Support for eBus by Valliant would be fine also :-) https://ebus.github.io/adapter/

jperquin, 2020-03-14:
Currently using Tado as homekit accessory / integration, but fed up with their cat & mouse approach API-wise, forcing our development community to play constant catch-up.

Have good wifi coverage throughout the house and a good deal of ESPHome devices that work well. Would want to switch to simple wifi controlled radiator thermostats and have HA manage all the smart logic, controlling the boiler / water heater thru opentherm.

Maybe someone can look at ihormelnik's Arduino code and port to ESPHome?

roma1141r1, 2020-09-17:
I subscribe to the question. Igor Melnik got a good, easily repeatable device and has open source codes, ideally this device should work in ESP HOME

Morphy99, 2020-09-24:
I had a look at ihormelnik's mqtt opentherm bridge but it looks like it only supports thermostat temperature. I'm looking for something that works with water temp and any other info the boiler might pump out

rsciriano, 2020-11-15:
Hi, I have integrated my OpenTherm boiler using  the [ihormelnyk's OpenTherm Adapter](http://ihormelnyk.com/opentherm_adapter) and custom components (you can see the code [here](https://gist.github.com/rsciriano/c78fc1e3f2d6afbb0295a2e5f4070174), it needs a good review/refactoring)

Now I would like to contribute to esphome, adding the appropriate components:

- OpenTherm Adapter
- OpenTherm Sensors
- OpenTherm Climates
- ??

I'll be commenting here on the progress

jperquin, 2020-11-15:
Very cool! Also from a perspective on how to adapt existing arduino projects to ESPHOME.

How do I organize the .h files in my ESPHOME setup so they get included in the firmware build process?

rsciriano, 2020-11-17:
Hi @jperquin, 

I had a few mistakes in my code _(It worked in my machine ;-)_

I have updated the gist, fixing these errors (opentherm_component.h and opentherm.yaml files)

You also need to use the dev version of esphome

good luck!

trueskalon, 2021-01-01:
Hi @rsciriano,

I trying to integrate esp8226 with your components, yaml files. Compiled, flashed, everything according to the "book" :). 

Now I am trying to integrate esp with HA with no luck :(. I tried 2 diffrent esp8266 - nodemcu and d1 lite pro.

The esp is connected to my router, it is "green" in esphome (dev) and from there I can upload bin via ota.

When I try to see logs (esphome) it shows:

```
INFO Reading configuration /config/esphome/opentherm.yaml...
INFO Starting log output from 192.168.100.58 using esphome API
INFO Connecting to 192.168.100.58:6053 (192.168.100.58)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 1 seconds
```

and its keep going... till:

```
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 30 seconds
INFO Connecting to 192.168.100.58:6053 (192.168.100.58)

```
 when i try to see logs via usb from esphome flasher:

`Unexpected error: could not open port 'COM3': PermissionError(13, 'Access denited.', None, 5)`

or other rubbish))

Cannot connect viat http too :(.

I try to use dr googl.
 I am no poweruser in this topic, yet :(.

Any help appricieted :)

trueskalon, 2021-01-05:
Hi,  

I have issues (as I can see from usb logs) with rebooting of my esp when trying **rsciriano** HA integration via esphome dev.

I tried esp8226 - d1 mini pro v1 and nodemcu v0.9 - no luck, same behivoure. 

I was flashing, earsing, new HA, few psu for powering, used caps on 3,3v and 5v rail and always got same result.

The esp is connected to my router I can do OTA updateds...

Any suggestions?

esphome log:
`WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 1 seconds`

usb log:
`l ¬[23:27:37] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[23:27:37]
[23:27:37]load 0x4010f000, len 3456, room 16 
[23:27:37]tail 0
[23:27:37]chksum 0x84
[23:27:37]csum 0x84
[23:27:37]va5432625
[23:27:42]@cp:0
[23:27:42]ld
`

rsciriano, 2021-01-13:
Hi @trueskalon,

I'm currently busy with other things so I can't get time to try to reproduce the error. I would also like to work on integrating this code into ESPHome :(

I don't know what could be happening, that same code is working in my boiler, in a Wemos D1 mini

Things that might help diagnose the problem:

- Use the stable version of ESPHome *(that's what I'm using)* 
- Connect the device by the usb station and load it into firmware using `esphome run ...`, to see the logs from the beginning
- Modify the implementation of `OpenthermComponent`, increasing in time for each loop, on 5 seg per sample *(If the communication arrives too long it could block the program and cause the error)* 
```
OpenthermComponent(): PollingComponent(5000) {
```
- Make sure the OpenTherm Adapter board is properly connected *(if the correct signals are not received, the program may also crash)*

I hope it will be helpful

wichers, 2021-01-22:
I'm obsessed in getting all my equipment to run esphome. I'm currently working on an OpenTherm gateway: 
https://github.com/wichers/esphome-opentherm

Current version is working, but work in progress. My plan is to incorporate the same features as this pic16f project: http://otgw.tclcode.com/firmware.html#dataids

As this is a native (ESPxxx RX/TX legs connected to Boiler and Thermostat) OpenTherm implementation you need hardware directly connected to your esp, for example:
https://github.com/jpraus/arduino-opentherm (I'm using this on a WeMos R32)
https://diyless.com/product/arduino-opentherm-gateway
http://ihormelnyk.com/opentherm_adapter

Weissnix4711, 2021-01-23:
Interesting stuff.

I have been trying to do something similar to @rsciriano for quite some time. Sadly, cpp isn't my strong point.

I've also seen [this library](https://github.com/jpraus/arduino-opentherm) which may be an alternative to Ihor Melnyk's library. The master interface of @jpruas' shield is virtually identical to Melnyk's adaptor, so this should be compatible.

diyless, 2021-02-21:
@wichers, Ihor has updated his OpenTherm library, now it supports DHW, just FYI

Weissnix4711, 2021-02-22:
> @wichers, Ihor has updated his OpenTherm library, now it supports DHW, just FYI

This is only some clean wrapper function, there is no need to add support for dhw, it's practically identical to CH, only a different msg id.

Weissnix4711, 2021-02-22:
As I understand, the main issue is not blocking the main loop for too long. Each frame is 32 bits, with 1ms for each bit. That may or may not be problematic, I am unsure quite how long we could block for.

ihormelnyk, 2021-02-22:
There is a sample at the bottom of the article how to use asynchronous requests http://ihormelnyk.com/opentherm_library
Also, there is a tutorial on how to build/send any of OT commands using my OpenTherm Library

serhat-ozkara, 2021-03-14:
Any progress on this? I would REALLY love to have my boiler working with PID on Opentherm. I don't know how commercial OT thermostats regulate but PID would be fine on boiler temp i guess. Currently on my setup, PID duty is handled on Node-RED and Tasmota with OT support. Unfortunately, tasmota thermostat implementation only works for on/off operation.

trueskalon, 2021-03-14:
@serhat, you can use rsciriano OT integration for controlling heating water temp via call service node, so you can create some kind of "heating curve" (you will need outside temp reading).

Just curious - what do you what to regulate (and read) with PID valves controls and OT boiler ? you want to include reading of return temp? regulate temp of heating water / outside temp and room temp and include it in PID function?

serhat-ozkara, 2021-03-14:
@trueskalon I use a bt sensor to read room temp & feed it to PID to change boiler's opentherm ot_boiler target temp. I then limit & map output of PID 0-100 to boiler temp to 30 - 55 range (which is economical, efficient). On/off control is inefficient as my boiler supports modulation of flame and on/off defeats that capability.  I have to do this because boiler central heating water temp 30 does not mean room temperature will be 30 degrees...
[image]

rsciriano, 2021-03-14:
Very interesting @serhat-ozkara!!!

I will have to incorporate PID to my code and convert the heating watter climate on an ambient temperature climate.

I have been doing some research and in [this post](http://ihormelnyk.com/opentherm_thermostat) by @ihormelnyk there is an implementation of a PID in C code. I have also located [this library](https://github.com/br3ttb/Arduino-PID-Library) that looks good.

I am currently cannot spend time on this, I will keep you informed if I have any progress :-)

serhat-ozkara, 2021-03-14:
@rsciriano You'll be surprised of the temp stability and comfort of my current setup. I'm providing a graph screenshot. But nodered adds another failure point.
There's already a PID thermostat entity along with autotune function on ESPHome:
https://esphome.io/components/climate/pid.html
May be that can be utilized?

[image]

rsciriano, 2021-03-14:
@serhat-ozkara Yes, [PID Climate Component](https://esphome.io/components/climate/pid.html) could be used :-)

In my code, the heatting watter climate should be replaced by a float out with the value of the PID

Let's see when I can dedicate some time to it. Also, I'm going to be away from home for a while so I can't test it :(

serhat-ozkara, 2021-03-15:
@rsciriano I'd gladly test it out. :)

MSMikl, 2021-03-21:
> Hi,
> 
> I have issues (as I can see from usb logs) with connecting to HA of my esp when trying **rsciriano** HA integration via esphome dev.
> 
> I tried esp8226 - d1 mini pro v1 and nodemcu v0.9 - no luck, same behivoure.
> 
> I was flashing, earsing, new HA, few psu for powering, used caps on 3,3v and 5v rail and always got same result.
> 
> The esp is connected to my router I can do OTA updateds...
> 
> Any suggestions?
> 
> Thank you :)
> 
> esphome log:
> `WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 1 seconds`
> 
> usb log:
> 
> ```
> ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3456, room 16 
> tail 0
> chksum 0x84
> csum 0x84
> va5432625
> @cp:0
> ld
> ```

Hi.
I've got the same issue and I solved it.
You should change pin numbers in component.h, because D1 mini lite and D1 mini have different pins. I use D1 mini and set input pin to 4 and output - to 5.
And it only works if it can read OT data, so you will see complete log inly when plug ESP into the shield.

It works nice, but my boiler doesn't send return temereature and pressure, so I've removed them

rsciriano, 2021-03-28:
Hi @serhat-ozkara, I've been working on integrating my code with PID Climate, and I've made progress :-)

You can see the code in my repo https://github.com/rsciriano/ESPHome-OpenTherm

I have tested by connecting it to Home Assistant (simulating a temperature sensor and without connection to the boiler) and it seems to work _(though it needs more work)_

[image]

serhat-ozkara, 2021-03-28:
That's  great news @rsciriano ! I can give it a try as i have another d1 mini lying around. Can you provide an example yml for esphome? I'm totally new to esphome.

rsciriano, 2021-03-28:
@serhat-ozkara, in my repo you have an example yaml, for try it

- clone my repo (https://github.com/rsciriano/ESPHome-OpenTherm)
- create a secrets.yaml with your wifi name and password
```
wifi_name: "mywifi"
wifi_password: "my password"
```
- execute the command `esphome opentherm yaml run`

If you have more questions, you can open an issue in my repo (to avoid making noise here).

Weissnix4711, 2021-04-10:
@serhat-ozkara I might suggest using weather compensation. Or even a combination of weather compensation + PID?

Your flow temperature should be calculated from the outside temperature using a weather compensation curve. On a warmer day you will not need to use a flow temperature as on a cooler day.

For example, [here](https://i.ytimg.com/vi/chelwjUyMWw/maxresdefault.jpg) (and the equation [here](https://gist.github.com/Weissnix4711/60330176884b8c07db2f18d7538d2d8e)) is a somewhat complex example of such a weather compensation curve (from Viessmann). `vtsoll` = boiler temperature, `rtsoll` = room temperature setpoint, `at` = outside temperature. `Niv` and `nei` are values used to modify the curve, though for the purpose of illustration they can be ignored. The first offsets the boiler temperature, and the second is used to modify the slope of the curve.

Weissnix4711, 2021-04-10:
I should also mention me and rspaargaren had a conversation on discord a while ago. He managed to come up with [this](https://github.com/rspaargaren/espopentherm/blob/main/main.cpp), sending messages via interrupts.

This is a step further towards an actual component as part of esphome. Following the contributing standards, any libraries should be re implemented if possible, and large(-ish) delays, such as that of sending an opentherm message synchronously, should be avoided. Sure, the custom component by @rsciriano does *work*, but what I am trying to say is it wouldn't get implemented as-is.

sergeybuglak, 2021-11-11:
@rsciriano 
Do you have plans on moving to async calls?

Weissnix4711, 2021-11-11:
> Now I think, that MQTT is not so bad, because MQTT stores all parameters and sends them to ESP even after reboot. And in my current implementation parameters aren't saving on ESP, it uses default parameters after reboot

MQTT also has its issues.

And it is possible to store values in flash, so they can be restored after reboot. Even if there are network issues, the ESP should never lose track of these params/states. Btw a nice trick if any components don't explicitly support recovering flash, you can always use a combination of global variables and scripts which run at startup.

> Something like https://esphome.io/components/sensor/homeassistant.html ?

@MSMikl was referring to MQTT's ['retain' functionality](https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/), where the MQTT broker (such as the MQTT addon for HASS) will remember the last state, and resend that state whenever any clients subscribe).

Weissnix4711, 2021-11-11:
> any progress on this issue? would love to see this implemented in esphome. Right now, I'm using a diyless example with mqtt.

For an implementation to be accepted, it must followed some quite strict guidelines. The hardest to figure out is probably going to be timings. Something could probably be thrown together using timer interrupts, but I don't know shit about how those work, and I don't have the time to figure it out. Sadly, due to the nature of the protocol, using existing serial implementations and abstractions probably won't work.

I'd suggest you join the conversation on the discord if you aren't already.

jorisa, 2021-12-19:
After some tinkering I am currently using [ESPHome-OpenTherm](https://github.com/rsciriano/ESPHome-OpenTherm), and it appears to be working great so far! I agree having this supported natively would reduce the barrier to entry. I have not bothered getting the MQTT solution to work, but I can't imagine it being less complicated than rsciriano's implementation.

Mio3000, 2023-02-22:
Hi all, I would also add to this request also the OpenTherm Gateway, im using the one from DIYLESS using Wichers repo https://github.com/wichers/esphome-opentherm, i added some stuff to it, but i have no clue how to change values for the setpoints like max ch water setpoint, and also i would like to have also climate option for the DHW temperature.

tricker1234, 2023-03-12:
Hi group! 
Had someone an error during instalation? 

"In file included from src/main.cpp:51:src/opentherm_component.h:3:10: fatal error: OpenTherm.h: No such file or directory"

I copied 0.5 version to config->esphome-> (there is opentherm.yaml and esphome-opentherm)

SebuZet, 2023-10-24:
> @serhat-ozkara I might suggest using weather compensation. Or even a combination of weather compensation + PID?
> 
> Your flow temperature should be calculated from the outside temperature using a weather compensation curve. On a warmer day you will not need to use a flow temperature as on a cooler day, therefore weather compensation should yield better results than PID alone.
> 
> For example, [here](https://i.ytimg.com/vi/chelwjUyMWw/maxresdefault.jpg) (and the equation [here](https://gist.github.com/Weissnix4711/60330176884b8c07db2f18d7538d2d8e)) is a somewhat complex example of such a weather compensation curve (from Viessmann). `vtsoll` = boiler temperature, `rtsoll` = room temperature setpoint, `at` = outside temperature. `Niv` and `nei` are values used to modify the curve, though for the purpose of illustration they can be ignored. The first offsets the boiler temperature, and the second is used to modify the slope of the curve, such as for underfloor heating you'd want to use a "flatter" slope than for radiators.

@Weissnix4711 thanks for the formula. I used it to create a controller for my boiler and underfloor heating system. I didn't use it wih PID controller. Do you think those to approaches can be combined?

Weissnix4711, 2023-10-24:
> @Weissnix4711 thanks for the formula. I used it to create a controller for my boiler and underfloor heating system. I didn't use it wih PID controller. Do you think those to approaches can be combined?

@SebuZet Yes, absolutely.

Using the example of Viessmann again (because that's what I know), they can run in 'witterungsgeführtem Betrieb
mit Raumeinfluss'. Ie. Weather compensated mode with room influence.

All that does is take the difference between setpoint and actual room temp, and multiplies it by a factor. This is then added (or subtracted) to the flow temp.

If you want to get really technical, this is the exact logic they use:

```
vtsoll = vtsoll + ((rtsoll - rtist) * (1 + nei) * (refaktor / 4))
```

`rtist` - room temp actual value (ist means 'is' in German)
`refaktor` - room influence (einfluss) factor. Default is 8.

The `nei` (slope) variable is just in there to make the setting feel more intuitive for any given curve. It doesn't really matter too much.

You'll notice this is literally just a P-only controller. I'm only posting the exact formula in case someone here has used Viessmann systems before, and might already have an intuition for what factor they need for their home. I'm not necessarily suggesting anyone implement this logic exactly.

However, what I've done is use a PI controller, where the input is the room temp error, and the output is summed with the output of the previous weather compensation logic. Similar to Viessmann's room influence logic. The derivative is a bit overkill and just another setting to mess with so I'd leave it. Importantly though, because this PID (or rather PI) controller is working in tandem with weather comp, its gonna be a lot more forgiving in configuration than if the entire system worked solely on PID.

Because of the very slow response time of the system, calibrating it and trying to prevent windup was/continues to be a  pain. But hey. It does help driving away any constant error due to inaccuracies in outside temp measurement (I'm just using the hourly weather forecast, which ain't perfect) and makes the system heat up much quicker.

kmarek1, 2023-12-03:
> thanks for the formula. I used it to create a controller for my boiler and underfloor heating system. I didn't use it wih PID controller.
Could you please share your solution?

ananyevgv, 2024-05-02:
https://github.com/FreeBear-nc/esphome-opentherm

ananyevgv, 2024-05-02:
https://github.com/khenderick/esphome-opentherm