esphome/feature-requests #522: TFT-LCD ST7789V of ESP32 TTGO

musk95, 2019-12-09:
**Describe the problem you have/What new integration you would like**

I'd like to ask to add TFT-LCD ST7789V of ESP32 TTGO.

**Please describe your use case for this integration and alternatives you've tried:**

This product is,
https://sites.google.com/site/jmaathuis/arduino/lilygo-ttgo-t-display-esp32

**Additional context**

There is for Arduino, but not exist for ESPHome.
so I create component for ESPHome and code is on https://github.com/musk95/esphome
Please include this at next release if it is possible.

Thank you.

brandond, 2019-12-09:
Can you put in a pull request?

musk95, 2019-12-10:
I've put in a pull request of the dev.
Thanks.

closed 2019-12-10 by musk95

minsuke, 2019-12-10:
thank you musk95 : )