esphome/feature-requests #523: Add Service to update Nextion Display on demand

carlos-sarmiento, 2019-12-10:
**Describe the problem you have/What new integration you would like**
A service to trigger refresh of a Nextion display

**Please describe your use case for this integration and alternatives you've tried:**
I have a nextion display that I don't need to update frequently (ideally every 30 seconds would be enough). My issue is that when a toggle is pressed, I want the display to update immediately (instead of waiting the 30s). Since there is no way to trigger a refresh, I'm stuck with a refresh rate of 1s that 99% of the time does nothing.

**Additional context**

OttoWinter, 2019-12-10:
See `component.update`

closed 2019-12-10 by OttoWinter

sergeolkhovik, 2019-12-10:
If Carlos is talking about next page feature, then I have such code:
```
globals:
  - id: timer_default
    type: int
    restore_value: no
    initial_value: '5'
  - id: timer_long
    type: int
    restore_value: no
    initial_value: '15'
  - id: timer_current
    type: int
    restore_value: no
    initial_value: '0'

binary_sensor:
  - platform: gpio
    id: button
    name: D1 Display Button
    internal: true
    pin:
      number: 14
      mode: INPUT_PULLUP
      inverted: True
    on_multi_click:
      - timing:
        - ON for at least 1.5s
        - OFF for at least 0.5s
        then:
          - lambda: |-
              id(timer_current) = id(timer_long);
          - display.page.show: page_indoor
          - component.update: Display
      - timing:
        - ON for at most 0.5s
        - OFF for at least 0.5s
        then:
          - lambda: |-
              id(timer_current) = id(timer_long);
          - display.page.show_next: Display
          - component.update: Display

interval:
    - interval: 1s
      then:
        - if:
            condition:
              lambda: |-
                id(timer_current) -= 1;
                if (id(timer_current) <= 0) {
                    id(timer_current) = id(timer_default);
                    return true;
                }
                return false;
            then:
              - display.page.show_next: Display
              - component.update: Display

```

Single press on button will switch to next page, long press will show first page for more long time than usual.