esphome/feature-requests #526: Multiple Home Assistant Areas

anthonyangel, 2019-12-15:
**Describe the problem you have/What new integration you would like**

I have several Sonoff 4ch's which control devices in several different rooms. The Sonoff devices are centrally located out of the way (in a cupboard and in the loft), but they control devices (lights & extractor fans) in other rooms. I'd like to use the Home Assistant Areas feature per-relay

**Please describe your use case for this integration and alternatives you've tried:**

I'd like to be able to set the Area per relay, not per ESP device

**Additional context**

My devices are connected via the ESP Home API

brandond, 2019-12-16:
This would be a Home Assistant feature request, wouldn't it? I believe the HA area code only lets you set a single area per device. ESPHome would need to do some sort of hack like creating virtual devices to allow you to split things in HA.

ng-galien, 2019-12-19:
@anthonyangel 
I have configurated several devices accross logical areas with separated phyical commands. 
To handle it in HA I use light groups with 2 to 3 level from single->group->all. The your output should be configurated as a light.
I only use these group in the HA frontend.
I use some inputs from physical button to command these groups. 
The logic is something like.

```
btn#0 click#0 -> group#0 ON
btn#0 click#1 -> group#1 ON
btn#0 click#2 -> group#2 ON

btn#0 long click -> group#ALL OFF
btn#1 double click -> group#ALL ON
```

In this way you can avoid every hardware coupling and simply reconfigure your installation. 
All the logic is done in Nodered and in the ESP for propagating input event.
In the ESP related to the button I update the state of each group with an API function called ny nodered to avoid not synchronized states.
(The code is quite uggly with custom C function in ESPHome, I will rewrite it in a clean comp but...)

deece, 2020-04-21:
I have the same issue with BLE sensors interfaced via ESPHome. I believe this request needs to be logged against the HomeAssistant ESPHome integration. Each entity should be exposed as a separate device, rather than as many entities under a single device. Those devices can then be assigned to rooms as required.

XtremeOwnageDotCom, 2020-06-29:
I have this issue as well. Using a shelly 2.5 to control the doubleswitch in my entryway. One switch- activate the porch lights (Front porch area). The other switch activates the lights in the Entryway (Hallway Area).

Also- I have ONVIF cameras. One inside of a bedroom for monitoring the kids, and one on the front porch. I can only assign both of the cameras to the same area, which is not what is expected.

munchj, 2020-10-13:
I have the same issue, one ESPHome to fetch the data of several LYWSD03MMC sensors, would love to have 1 device per entity.
Could maybe be solved with mqtt/nodered instead of using the api, but I just installed my stack and didn't fiddle with nodered yet.

antoni-k, 2020-10-18:
As @brandond said, I think that virtual devices created by the ESPHome integration and defined in the firmware configuration would be an optimal solution. Maybe something like creating a virtual device as a top-level entry, then adding sensors or whatever else under it?

oh2ftu, 2020-10-20:
> I have the same issue, one ESPHome to fetch the data of several LYWSD03MMC sensors, would love to have 1 device per entity.
> Could maybe be solved with mqtt/nodered instead of using the api, but I just installed my stack and didn't fiddle with nodered yet.

I've been looking into this exact issue for days, has there been any progress?

frenck, 2021-05-11:
This feature has actually been implemented on the Home Assistant side.

[image]

When looking at an entity, in Home Assistant, you can now change/override the area of a specific entity to be different than the device.