esphome/feature-requests #527: Micropython with ESPHOME

ng-galien, 2019-12-19:
**Describe the problem you have/What new integration you would like**

Use ESPHome native API with micropython on the ESP

**Please describe your use case for this integration and alternatives you've tried:**

For hobby stuff, I would like to use few ESPHome components with micropython on the ESP side, such as binaries and APIs functions.
The purpose of my request is to ask where to start. As I understand the protocol layer is in the aio repo, is it right? 
As I understand the architecture of ESPHome is a server on the device whose respond to request from HA.
Is there any document which describes it? My purpose is to known how to get started within the ESPHome code.

Thanks.

**Additional context**

lwqcz, 2019-12-26:
@ng-galien you mean something like this? https://esphome.io/components/api.html

ng-galien, 2019-12-28:
> @ng-galien you mean something like this? https://esphome.io/components/api.html

My original thought was to translate the esphome communication in micropython. Everything is in the protobuf definition.

But right now I'm exploring another way:
- Avoid the communication and UI layer, in fact HA
- Keep Node Red for communication / automation / UI
- Run micropython on the devices with very basic communication API with Node Red
 
The big advantage of micropython is that we can run our custom code without recompiling / uploading a new firmware, we just have to write a file and transfer it. I can be done dynamically too. 
My personal needs at home are at 100% controlling relays over digital inputs, that's why I can quickly move to that solution.
For now I can control DIO of an ESP8266 (Sonoff basic) from Node Red, with a small portion of code on the device, no MQQT or other middle ware are needed.

So, I will close that Feature request as it's out of the scope of ESPHome.

[image]

rudshep, 2020-01-05:
Before discovering ESPHome I was well on my way to put all my eggs in the Micropython basket using uPyEasy https://www.letscontrolit.com/forum/viewtopic.php?t=3577 and https://github.com/letscontrolit/uPyEasy . However for the majority of my projects, I require rather "dumb" nodes that simply flip switches and toggle relays so putting all the smarts (and weight) of MicroPython on the node itself doesn't make sense. The bigger problem is Micropython is fantastic at executing "rules" but not be the "Operating System" itself like uPyEasy tries to do. Some things are just better done in C and kudos to Otto for creating the ESPHome magic. 

However, I have use cases for "super smart nodes" where the node should be local and autonomous and not rely on HASS or NR to perform the complex logic. I too ran into situations where putting a complex Lambda routine on the ESPHome node is a painful process - especially the code/compile/upload/wait for connection/attach to log/look for error message/repeat cycle. Not to mention my lack of C++ skills to code something even moderately complex. 

A Micropython component that will execute a "Rule" defined in YAML and uploaded without needing to compile the project would be very cool - e.g. ESPHome solardiverter.yaml runrules. Perhaps there is a case to be made to use Micropython as the Rules engine for local ESPHome execution? It would require embedding the Micropython engine into ESPHome and only run on ESP32 or large memory devices but could be amazing.

ng-galien, 2020-01-05:
@rudshep

closed 2020-01-09 by ng-galien