esphome/feature-requests #528: Save configuration .yaml in compiled binary

prabbit237, 2019-12-19:
**Describe the problem you have/What new integration you would like**

The FAQ warns that the configuration .YAML file doesn't get stored in the binary but that would be a really NICE feature as it would make recovery a lot easier. 

**Please describe your use case for this integration and alternatives you've tried:**

Currently, if you lose the .yaml file used to create a binary, you're pretty much at the apex of the narrow, winding waterway without a flat manual-propulsion device as far as rebuilding the code goes.

**Additional context**

https://stackoverflow.com/questions/42235175/how-do-i-add-contents-of-text-file-as-a-section-in-an-elf-file shows a fairly easy way to compile a text file into an object file and have it linked in to the resulting binary. It's referencing X86/64 gcc compilers but at a glance, looks like it should work with any gcc compiler/toolchain that has the objcopy utility. I don't know how easy it would be to incorporate this into the platformio/ESPHome scripts, however.

glmnet, 2019-12-21:
Then there will be the need to implement some yaml retrieval mechanism which is some more work. 
May be adding an option to web_server to also embed and provide a link to the yaml can be an easy to implement one. 
I find this also useful to be aware at the right code running on the esp, to help in version control. 
I regularly push yaml to git so this is not in my worry list at all.

nagyrobi, 2022-07-07:
Please see https://github.com/esphome/feature-requests/issues/1507 there's an external component for that.
Duplicate of https://github.com/esphome/feature-requests/issues/1507

closed 2022-07-07 by nagyrobi