esphome/feature-requests #530: Modbus TCP (Textsensor Component)

OE5DRO, 2019-12-21:
**Describe the problem you have/What new integration you would like**
It would be nice to connect a modbus tcp client and read out some registers.

**Please describe your use case for this integration and alternatives you've tried:**
I would like to connect to my victron raspberry VRM and read for example the voltage
of my battery and diplay it on a esp module and switch something with a relay board if
voltage rises a to high or low value.

**Additional context**
I thing its not vendor dependend because modbustcp is universal.
I found a modbus integration with UART is already implemented but not with tcp connection

randybb, 2019-12-22:
I don't believe this will be implemented, since you can achieve this via home assistant - create a [modbus](https://www.home-assistant.io/integrations/modbus/) sensor on it and then read value via [homeassistant](https://esphome.io/components/sensor/homeassistant.html) sensor platform.

OE5DRO, 2019-12-23:
i don't use homeassistant, because i allready use homebridge.
maybe someone knows more about the existing modbus functionality in esphome.
its not documented, but known by the dashboardeditor.
if entered in yaml file it asks for speed and RX. TX pins and uart.
i think this integration is not far away from modbus tcp

glmnet, 2019-12-30:
having ESPHome act as a modbus slave or master via tcp would be awesome, even though it is meant to be used with HASS, ESPHome is becoming one of a kind platform, i like to call it 4th gen microcontroller language

rudshep, 2020-01-05:
From what I can see in the Modbus component source, it is meant to be used over UART (Modbus RTU) and Modbus TCP is not implemented yet. Modbus TCP is pretty much just the RTU (serial) comms encapsulated in a TCP packet so it shouldn't be too difficult to do. Being a master/slave architecture, you will still need a Master to monitor and control things - i.e. Home Assistant or Node Red or whatever and using another ESPHome node as a master is probably too much to ask. A Modbus TCP sensor and "coil" (actuator) component in ESPHome will be fantastic though.

SunsetGolfer, 2021-03-01:
Maybe a good basis to include it in esphome?

https://github.com/emelianov/modbus-esp8266

matthias882, 2022-04-17:
Modbus TCP-Support would be great. I would like to use it for reading out values from a power-meter and use this to controll some devices directly without any failure-possibilities in this line (iobroker, homeassistant or something else)