esphome/feature-requests #531: delayed_on_off: different on and off

Erelen-Laiquendi, 2019-12-22:
Thanks for delayed_on_off - https://github.com/esphome/esphome/pull/700

**Describe the problem you have/What new integration you would like**
Current syntax:
````
filters:
  - delayed_on_off: 40s
````
Please, add ability to define different on and off time. For example, like this:
````
filters:
  - delayed_on_off:
      on: 5s
      off: 30s
````

**Please describe your use case for this integration and alternatives you've tried:**
In one mode, my washing machine has power consumtion pattern like this (simplified for clear example):
> 10 seconds > 0W
> 20 seconds = 0W
> 10 seconds > 0W
> 20 seconds = 0W
> ...

And (!) from time to time there is very short random consumtion spikes, when machine don't work.

I want to make binary sensor "machine is on". So, I need short delay on filter (~5s) to ignore random spikes, but not more than 10s. And long (>20s) delay off filter.

As I understand (thanks to your explanation about pipeline - https://github.com/esphome/issues/issues/463), I can't use delay_on + delay_off (or I can't understand, how).

But something like this totally solve my case:
````
filters:
  - delayed_on_off:
      on: 5s
      off: 25s
````

Alternative solutions:
1. Send power consumtion to HA and create template binary sensor with delay_on and delay_off (it's pity - I can simple do it in HA, but can't in ESPHome).
2. In ESPHome create two sensors - one with delay_on and second with delay_off, and binary_sensor.template.publish from them to third binary sensor. Ugly :(

glmnet, 2019-12-30:
sounds like a job for the [multi click filter](https://esphome.io/components/binary_sensor/index.html#binary-sensor-on-multi-click)

nagyrobi, 2022-07-07:
Implemented separate `delayed_on` and `delayed_off`: https://esphome.io/components/binary_sensor/index.html#delayed-on

Erelen-Laiquendi, 2022-07-07:
Separate `delayed_on` and `delayed_off` is not the same.
That's explanation by [OttoWinter](https://github.com/OttoWinter): https://github.com/esphome/issues/issues/463#issuecomment-503959711
Please, reopen this issue.

SalexSun, 2023-01-05:
@Erelen-Laiquendi did you ever find a solution for this?

Erelen-Laiquendi, 2023-01-06:
Current ugly solution:

```yaml
binary_sensor:
  - platform: template
    id: activity_moment
    lambda: |-
      if (isnan(id(power).state)) {
        return {};
      } else if (id(power).state > 2.5) {
        return true;
      } else {
        return false;
      }

  - platform: template
    id: activity_on
    lambda: 'return id(activity_moment).state;'
    filters:
      - delayed_on: 2s
    on_press:
      - binary_sensor.template.publish:
          id: activity
          state: ON

  - platform: template
    id: activity_off
    lambda: 'return id(activity_moment).state;'
    filters:
      - delayed_off: 30s
    on_release:
      - binary_sensor.template.publish:
          id: activity
          state: OFF

  - platform: template
    id: activity
    name: "Washer Activity"
    icon: mdi:washing-machine
```

felixstorm, 2023-10-07:
@Erelen-Laiquendi I just found your issue searching for a solution for the same problem and by random discovered that it might have actually been implemented almost secretly a few months ago together with making delays templatable: https://github.com/esphome/esphome/pull/5029  
It also gets mentioned in the docs by now (`time_on` and `time_off`): https://esphome.io/components/binary_sensor/index.html#delayed-on-off

I have not had a chance to try it out yet, but from looking at the code it sounds very promising. What do you think?