esphome/feature-requests #532: Add binary_sensor functionality to tuya dimmers

lawadam, 2019-12-22:
**Describe the problem you have/What new integration you would like**

Being able to double / multi click would be helpful for tuya dimmers. For example, double clicking the main button on other dimmers usually sets the brightness to 100%.

**Please describe your use case for this integration and alternatives you've tried:**

I've used the default tuya dimmer component (which works quite well).

**Additional context**

I'm using the feit dimmers from costco (us). I can send logs / take pictures of internals if that is helpful.

brandond, 2019-12-23:
Yeah, if you could post verbose log output from your device that'd be great. I have some additional Tuya stuff in the pipeline, and it's nice to know what kind of hardware people have.

lawadam, 2019-12-23:
Push main button (turn on):

> [11:58:15][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=3 DATA=[01.01.00.01.01 (5)] INIT_STATE=5
> [11:58:15][V][tuya:235]: Datapoint 1 update to 1
> [11:58:15][D][light:264]: 'light_family_overhead dimmer' Setting:
> [11:58:15][D][light:273]:   State: ON
> [11:58:15][V][tuya:275]: Datapoint 2 set to 172
> [11:58:15][V][tuya:279]: Not sending unchanged value
> [11:58:15][V][tuya:275]: Datapoint 1 set to 1
> [11:58:15][V][tuya:279]: Not sending unchanged value

Push brightness increase button (twice):

> [11:59:51][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=3 DATA=[02.02.00.04.00.00.00.C8 (8)] INIT_STATE=5
> [11:59:51][V][tuya:235]: Datapoint 2 update to 200
> [11:59:51][D][light:264]: 'light_family_overhead dimmer' Setting:
> [11:59:51][D][light:277]:   Brightness: 20%
> [11:59:51][V][tuya:275]: Datapoint 2 set to 200
> [11:59:51][V][tuya:279]: Not sending unchanged value
> [11:59:51][V][tuya:275]: Datapoint 1 set to 1
> [11:59:51][V][tuya:279]: Not sending unchanged value
> [11:59:51][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=3 DATA=[02.02.00.04.00.00.01.5E (8)] INIT_STATE=5
> [11:59:51][V][tuya:235]: Datapoint 2 update to 350
> [11:59:51][D][light:264]: 'light_family_overhead dimmer' Setting:
> [11:59:51][D][light:277]:   Brightness: 35%
> [11:59:51][V][tuya:275]: Datapoint 2 set to 350
> [11:59:51][V][tuya:279]: Not sending unchanged value
> [11:59:51][V][tuya:275]: Datapoint 1 set to 1
> [11:59:51][V][tuya:279]: Not sending unchanged value

Push brightness decrease button (twice):

> [11:59:52][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=3 DATA=[02.02.00.04.00.00.00.C8 (8)] INIT_STATE=5
> [11:59:52][V][tuya:235]: Datapoint 2 update to 200
> [11:59:52][D][light:264]: 'light_family_overhead dimmer' Setting:
> [11:59:52][D][light:277]:   Brightness: 20%
> [11:59:52][V][tuya:275]: Datapoint 2 set to 200
> [11:59:52][V][tuya:279]: Not sending unchanged value
> [11:59:52][V][tuya:275]: Datapoint 1 set to 1
> [11:59:52][V][tuya:279]: Not sending unchanged value
> [11:59:52][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=3 DATA=[02.02.00.04.00.00.00.32 (8)] INIT_STATE=5
> [11:59:52][V][tuya:235]: Datapoint 2 update to 50
> [11:59:52][D][light:264]: 'light_family_overhead dimmer' Setting:
> [11:59:52][D][light:277]:   Brightness: 5%
> [11:59:52][V][tuya:275]: Datapoint 2 set to 50
> [11:59:52][V][tuya:279]: Not sending unchanged value
> [11:59:52][V][tuya:275]: Datapoint 1 set to 1
> [11:59:52][V][tuya:279]: Not sending unchanged value

Push main button (turn off):

> [12:05:02][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=3 DATA=[01.01.00.01.00 (5)] INIT_STATE=5
> [12:05:02][V][tuya:235]: Datapoint 1 update to 0
> [12:05:02][D][light:264]: 'light_family_overhead dimmer' Setting:
> [12:05:02][D][light:273]:   State: OFF
> [12:05:02][V][tuya:275]: Datapoint 1 set to 0
> [12:05:02][V][tuya:279]: Not sending unchanged value

The repeating heartbeat:

> [12:05:06][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
> [12:05:06][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=3 DATA=[01 (1)] INIT_STATE=5
> [12:05:06][V][tuya:115]: MCU Heartbeat (0x01)