esphome/feature-requests #541: Allow to add existing Esphome nodes to a new Hassio instance for future OTA updates

MimbaMonkeyHouse, 2019-12-29:
**Describe the problem you have/What new integration you would like**

Allow to add existing Esphome nodes to a new Hassio instance for future OTA updates. 
**Please describe your use case for this integration and alternatives you've tried:**

I understand that the yaml code can not be recovered this way. However, it would help to be able to add existing nodes created with an old Hassio instance to a new instance without having to plug the ESP's via USB again.
**Additional context**

I’ve got 5 nodes running that were set up with a Hassio instance that had its microSD card corrupted and migrated my Hass to a new Pi 4. Right now, the only way to regain control of the ESP nodes is to plug them via USB to the new Hassio instance.
This is impractical/impossible as some are tucked away in walls or ceilings and will need to have sensors unsoldered before being able to move them to a new location. Bringing the new Hassio device closer is not always possible as ESP nodes have been made waterproof with cases and hot glue...

szurr, 2019-12-29:
Hi I made it dirty way simple copy esphome folder from backup file(unpack with 7zip). Maybe it was done after recover add-on from backup. But it works

glmnet, 2019-12-30:
As @szurr  mentioned, the ESPHome dashboard will understand your nodes are there and will try to OTA to them, ESPHome is fully backup-able via integrated backup mechanism of HA.

The only thing that ESPHome uses to know the nodes are there is the hostname or IP address if you are using fixed ip. So if the node is there, OTA should work.

I really don't know what this FR is for, retrieving YAML from nodes is another FR that might help this

closed 2019-12-30 by glmnet

MimbaMonkeyHouse, 2020-01-06:
I have no backup for the existing nodes. These still work and I can connect to their webservers and upload a new  firmware from there. I can however not OTA or see their status from ESPHome. Is this not a normal behavior then?

glmnet, 2020-01-06:
The thing is, ESPHome dashboard will not start tracking nodes after you upload to them, it should do it just because the node files are there, so try creating the file with the right node name.
It should show up online on your dashboard, if you cannot get it, I can to reproduce it myself here.

If you want to OTA to them, you should be able to do it, so long the config you downloaded to them previously had the `ota:` option set properly.

MimbaMonkeyHouse, 2020-01-06:
Hey Guillermo,
Thanks for your time. I will have a look in the morning. If I understand what you suggest correctly, it should be quite easy. 

Will need more frequent backups now that I know how easily corrupted micro sd cards get on pi...