esphome/feature-requests #543: Add Idle State Option for Remote Transmitter Configuration

AdamKGoddard, 2019-12-31:
**Describe the problem you have/What new integration you would like**

For some configurations of IR Emitter circuits (ie. the high power output of the Xiaomi Mijia IR Controller - ESP32 Single Core), a certain idle state is required (in this case high), otherwise the transistors and LEDs remain on and heat the board.

**Please describe your use case for this integration and alternatives you've tried:**

Setting the output inverted prevents the board from getting hot, but also sends incorrect IR signals. Tested with devices and with AnalisIR.

Setting up the output (Xiaomi GPIO26) as a GPIO switch results in:
 * ON / High > Transistors Off / Cool.
 * OFF / Low > Transistors On / Hot.

Adding something like "idle_state: high"/"idle_state: low" to the "remote_transmitter" configuration options and then using these states when not sending IR would solve this issue.

**Additional context**

Current Yaml file used for testing:
https://github.com/AdamKGoddard/XiaomiIR_ESPHome/blob/master/xiaomi_ir.yaml

glmnet, 2020-01-06:
Not so elegant but, did you try to switch off the transmitter after transmitting?

I mean, create an output (like the power led)

then add the turn "off" (actually on)

````yaml
  
output:
  - platform: gpio
    pin: GPIO23     #Blue LED - Powered on always
    id: PowerLED
  - platform: gpio
    pin: GPIO26  #Same as transmitter output
    id: TransmitterLED

switch:
  - platform: template
    name: ${upper_devicename} LG TV Power On
    turn_on_action:
      - remote_transmitter.transmit_lg:
          data: 0x02FD48B7              
          nbits: 32
          transmitter_id: internal
      - output.turn_on: TransmitterLED
````

I'm wondering if this shouldn't be flagged as a bug instead of a FR, if it's inverted may be the expected behavior would be to leave the output high

Please send feedback of this.

AdamKGoddard, 2020-01-06:
I tried manually turning off as a separate switch, but not as part of the template. Will try it out when I return home. You're right though, because I want to be able to have the IR as just an endpoint service through the API or MQTT, IR codes should be sent and read from Home Assistant.

AdamKGoddard, 2020-01-12:
Confirmed, does not fix the issue. Setting as a standard GPIO sets the output low on startup. Setting as an inverted output starts the board heating as soon as power is applied, and has the effect of also inverting the IR transmitted codes. Setting as a standard GPIO and setting "on_boot" to send high stops the board heating as soon as power is applied, but as soon as a code is sent, heating re-occurs. Sounds like an issue with the GPIO resource allocation?