esphome/feature-requests #544: Allow MQTT QOS for publish state

iharris, 2020-01-01:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nmeachen, 2022-09-17:
Not being able to set QOS makes automations flakey on devices at the edge of WiFi coverage. If a device drops of the network briefly at the time it is supposed to turn on or off, the state change is missed. QOS 2 is a must in those scenarios.

tjsaunders, 2022-10-22:
For information, the MQTT client used by ESPHome (paho.mqtt.client), (now?) supports QOS, meaning that changing this to be supported by ESPHome looks like it would just involve adding the configuration option, and pulling that configuration option through to be included in the publish method in each of the different MQTT sensor components.

That said, beyond my capability to do it properly, although it no longer requires changes to 3rd party libraries.

jenkinsdsrk, 2023-07-11:
Following up after almost 8 months. I don't currently have the know-how or time to get the know-how, but would like to use the QOS feature. Is there any update on this?

jayge91, 2023-08-05:
I am also wondering if there has been any movement. QoS 0 as default with no option to change seems like a missed opportunity.
I'd offer to work this myself but I have no clue where to start

alessandroBalliniGPA, 2024-01-11:
The question is, WHY when you implement such feature you decide to set default QOS to zero!

The mininum should be 1