esphome/feature-requests #547: Identify which GPIO cuased awake from deep sleep of ESP32 ext1

rklueber, 2020-01-03:
**Describe the problem you have/What new integration you would like**

It is possible to identify which PGIO caused the wake up from the method

    esp_sleep_get_wakeup_cause()

My feature request is to access this information from within ESPHome.

You can find code for an example sketch here:
https://randomnerdtutorials.com/esp32-external-wake-up-deep-sleep/

**Please describe your use case for this integration and alternatives you've tried:**

I am working on getting my post box smart to tell me I have mail. However no power in the postbox. Thats why I go with deep sleep. I have to reed contacts in my postbox which wake up the ESP32.

He goes into WLAN/MQTT and posts I_HAVE_MAIL depending which contact was triggered. Upper flap (post man uses this one) sets the flag, lower flap (the one to empty the post box) resets the flag.

Easy. Less than 80 lines of yaml with ESPHome (did I say I was impressed ,-).

Currently I need a flap/reed to be open at least 2sec. This is the time until the ESP is booted, logged into Wifi/MQTT. It works, but I have some false negatives when the postman is fast putting my mail in. If I would be able to identify the GPIO causing the wake up the lag would be zero.

**Additional context**

Great work here with ESPHome. I am so impressed. I was into ESPs some month ago but refrained from getting into using them more widely in my smart home (built 2007; mainly KNX and hue) because of the hassle in programming. Recently I went into christmas lights hobby and after WLED i found ESPHome. With ESPHome using those ESPs gets so much easier. I am sooo impressed.

supermanchris4, 2020-01-13:
+1
Without access to esp_sleep_get_wakeup_cause() how else can you tell the difference between a timer wake up or a pin wake up?

jappish84, 2020-02-03:
Could this be related to the possibility of using a touch pin for wake up aswell? I've experimented a bit with touch pin wake up from deep sleep, but i can't get the esp32 to stay in deep sleep with touch pin set as wakeup pin, my guess is because the pin is floating and esphome isn't handling that correctly?

timo-merlin, 2020-02-14:
This would be great for building battery powered remotes. +1

darylsargent, 2020-04-01:
I have a similar use case to the OP, except want to report battery level on timer wakeup and mail arrival on pin wakeup.  May code this manually for the moment but would prefer it to be in a managed ESPhome solution.

supermanchris4, 2020-04-11:
I’ve cracked it! I have been trying to get this working on and off for about 6 months. Thanks drogfild
& truglodite on https://community.home-assistant.io/t/deep-sleep-and-multiple-buttons/167214/11 for pointing me in the right direction.

The solution was to use a Global variable in the on_boot and add a sensor template that returns the Global variable. I also ended up using a priority of 900 but haven’t tried changing it to see if it makes a difference.

I’m using ext0 so use esp_sleep_get_wakeup_cause() but it should work for ext1 with esp_sleep_get_ext1_wakeup_status()

My configuration is:
```
  on_boot:
    priority: 900
    then:
      lambda: |-
        id(wake_up_reason) = esp_sleep_get_wakeup_cause();

deep_sleep:
  run_duration: 2min
  sleep_duration:
    days: 1
  wakeup_pin:
    number: GPIO32
    inverted: True
    mode: INPUT_PULLUP
  wakeup_pin_mode: KEEP_AWAKE
  id: deep_sleep_1

globals:
   - id: wake_up_reason
     type: int
     restore_value: no
     initial_value: '0'

sensor:
  - platform: template
    name: "Mail Wake Reason"
    accuracy_decimals: 0
    lambda: |-
      return (id(wake_up_reason));

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO32
      inverted: True
      mode: INPUT_PULLUP
    name: "Mail Notification"
    id: mail_notification
    device_class: occupancy

```

closed 2022-07-01 by nagyrobi