esphome/feature-requests #551: Support for Philips BLE nightlight (MUE4094RT)

vevsvevs, 2020-01-04:
**Describe the problem you have/What new integration you would like**

New Philips BLE nightlight with motion detection came to the market, so it would be really nice to have this device being used with the BLE-tracker platform. It sends only one event when light is turned on. [Here is VERY_VERBOSE log](https://pastebin.com/1NvuywJr).

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

vevsvevs, 2020-01-13:
Ok, made it by myself, [here](https://github.com/esphome/esphome/pull/951) is PR :)

ahpohl, 2020-05-03:
@vevsvevs I incorporated your [PR#951](https://github.com/esphome/esphome/pull/951) as best as I could into my current patch [ahpohl/esphome:xiaomi_complete](https://github.com/ahpohl/esphome/tree/xiaomi_complete).

For my understanding, the nightlight sends a motion detected signal once activated by a person walking by. Every so often the status is broadcasted via BLE. The motion detected state is pushed to HA if the motion has been detected for longer than the specified timeout period, otherwise a no motion detected signal will be sent. Please correct me if I am not understanding the device and your patch correctly. 

I use the following config:
```
binary_sensor:
  - platform: xiaomi_mue4094rt
    name: "MUE4094RT Nightlight"
    mac_address: "7A:80:8E:19:36:BA"
    timeout: "5s"
```
But I get a platform not found error:
```
Failed config

binary_sensor.xiaomi_mue4094rt: [source development.yaml:131]
  
  Platform not found: 'binary_sensor.xiaomi_mue4094rt'.
  platform: xiaomi_mue4094rt
  name: MUE4094RT Nightlight
  mac_address: 7A:80:8E:19:36:BA
  timeout: 5s
```
Any ideas?

ahpohl, 2020-05-03:
I forgot to rename `sensor.py` to `binary_sensor.py`. The component compiles now but I cannot test it as I do not have such a device.

no0ner, 2020-10-02:
I own 3 MUE4094RT but can't seem to get them working in home assistant. I really don't know where to start. anyone willing to help?

nagyrobi, 2022-07-06:
Implemented in PR above

closed 2022-07-06 by nagyrobi