esphome/feature-requests #553: Add device_class to sensors.

zewelor, 2020-01-05:
**Describe the problem you have/What new integration you would like**
I would like to sensors to declare device_type attribute. For example temperature sensors ( from any device currently ? ), only registers icon mdi:temperature. It would be good to or / and declare "device_class: temperature". Looks like its current approach in HA, used to check sensor type, and act on it somehow. 

**Please describe your use case for this integration and alternatives you've tried:**
My current use case is ability to auto export temperature sensors, to Google Assistant

**Additional context**
Example of temperature sensor, view in HA, wich device_class. 

https://github.com/home-assistant/home-assistant/pull/24264#issuecomment-570898457

Current esphome temperature sensor ( using bme280 ). 

[image]

glmnet, 2020-01-10:
I this can be done already by [customizing the entity](https://www.home-assistant.io/docs/configuration/customizing-devices/) in HASS, I understand it will be more convenient to define the attribute in ESPHome but just to make a fair issue.

glmnet, 2020-01-10:
This is tightly related to https://github.com/esphome/feature-requests/issues/360 and will aid in an effort to make ESPHome sensors / binary_sensors more consistent.

sermayoral, 2020-06-16:
Same request here. I have lots of battery sensors without the 'device_class' set to 'battery'. There are some custom components to check the battery level of all your devices. The battery sensors defined in ESPHome don't work.

@glmnet i have done customizing the entity, but you have to remember doing it every time, so it quite more convenient doing it in ESPHome.

akoivist, 2020-09-21:
Same here. I circumvented this using Nodered: changed mqtt/discovery_prefix to a custom one, then subscribed to that in Nodered, which then adds the device_class attribute (and also changes device attributes to get separate devices for all the BLE sensors), and then publishes to the homeassistant/sensor-topic.
Really would like to do this in ESPHome.

jseidl, 2020-09-28:
> 
> 
> I this can be done already by [customizing the entity](https://www.home-assistant.io/docs/configuration/customizing-devices/) in HASS, I understand it will be more convenient to define the attribute in ESPHome but just to make a fair issue.

One limitation is that customizations might not be available for custom_components at load time (my case). Looking forward for having `device_class` natively from ESPHome

ogiewon, 2021-02-07:
I would also like to see device_class be added to sensors.

closed 2021-02-15 by glmnet

mschae, 2021-03-08:
Hey not sure if I should open a separate issue or if this is related enough to re-open this issue: The `device_class` is not reported via MQTT autodiscovery.

glmnet, 2021-03-08:
Open a new issue please