esphome/feature-requests #555: M5Stick  sh1107 led-driver

Nyno-Kun, 2020-01-07:
**Describe the problem you have/What new integration you would like**

I am currently trying to print text to my M5Sticks (grey) and i cannot get this done because (i at least think) there is only  sh1106 drivers but none for  sh1107 that would be used

**Please describe your use case for this integration and alternatives you've tried:**

I've tried all possible setups for SSD1306

**Additional context**

petosak, 2020-08-04:
I'd like to use M5 Stick with SH1107.

SeNike, 2020-09-07:
I'd like use M5 Stick with SH1107 too.

comatose-tortoise, 2021-04-08:
I'd like to use the featherwing oled board, which is a SH1107.

pauldogg, 2021-11-18:
+Yes please.

pauldogg, 2021-11-20:
I don't know how to code, but for those who do, perhaps this helps?

https://www.espruino.com/SH1107

arunderwood, 2022-01-01:
I was able to get the [Adafruit FeatherWing OLED - 128x64 OLED](https://www.adafruit.com/product/4650) working with some minor tweaks to the `SSD1306` component, I've pushed my code up here https://github.com/esphome/esphome/pull/2967.  The three changes I had to make can be summarized as:

* Width and Height were rotated from what I expected, it's a "vertical" display.
* The [VCOM value](https://github.com/esphome/esphome/pull/2967/files#diff-5ff3e32d63004d5493a0327c5470142268a9429418915b2ee5a38b038feb2b21R117) was different than the existing displays
* In the YAML config, the output needed to be rotated and offset in order to fit within the display

Figuring these things out was a mix of trial and error and referencing the commands from the [Adafruit_SH110x driver](https://github.com/adafruit/Adafruit_SH110x/blob/af992ff9da10bb6a55d22d1d40d0d5c25d33593b/Adafruit_SH1107.cpp)

randybb, 2022-01-01:
Just tried it with https://shop.m5stack.com/products/oled-unit-1-3-128-64-display and it doesn't work - display is showing random dots only.

arunderwood, 2022-01-02:
I haven't looked at the m5 example code, but it's possible that the m5 display is horizontally oriented instead of vertical.  I believe this could be tested by inverting height and width [here](https://github.com/esphome/esphome/pull/2967/files#diff-5ff3e32d63004d5493a0327c5470142268a9429418915b2ee5a38b038feb2b21R207-R246)

closed 2022-07-01 by nagyrobi