esphome/feature-requests #556: BLE GATT API

torfbolt, 2020-01-08:
**Describe the problem you have/What new integration you would like**

A general BLE GATT API in esphome (for ESP32 boards).

**Please describe your use case for this integration and alternatives you've tried:**

I think a lot of new devices could be supported quite easily if there was a simple and well designed API to send & receive from GATT endpoints.

**Additional context**

This is the closest I found, maybe it would be a good starting point:
https://github.com/ayufan/esphome-components/tree/master/esp32_ble_clients
However, from looking at the two modules using it in the same repo I have the feeling that the API could be made simpler/cleaner/more documented.

askthisthing, 2020-01-22:
Sending GATT commands to a specific BLE MAC addresses would indeed be awesome!

ralle12345, 2020-01-27:
I don''t know it it fits into this feature-request, but I think I could solve one problem when receiving from GATT endpoints was implemented. 

I use bluetooth tags on the keyring for presence detection. I would like to read the battery status of the tags every once in a while and warn if the battery should be changed.

skis4hire2, 2020-02-15:
Here is an esp32 BLE to mqtt standalone project. 
https://github.com/shmuelzon/esp32-ble2mqtt

It would be nice to have similar GATT exposure through the native API.

Also, if this were implemented, would it be possible to run this code and also the BLE tracker component at the same time?

ralle12345, 2020-02-16:
I looked at https://github.com/shmuelzon/esp32-ble2mqtt as well and it seems that the GATT functions would need to connect, get values and then disconnect again to keep the device broadcasting it's beacon frames.

jjok, 2020-06-02:
I was just going to try out esp32-ble2mqtt, but I don't really understand how to. It would be great if ESPHome had support for generic GATT characteristics.

2m, 2021-03-14:
Having this would allow to add Valve Index Base Station status and control support to Home Assistant.

koenvervloesem, 2021-03-15:
There's a pull request for a BLE client: https://github.com/esphome/esphome/pull/1177

dkisselev, 2021-05-20:
The pull request has been merged and functionality has been released in 1.18.0 🎉

jhansche, 2021-05-20:
The mentioned PR is merged, but there are some limitations:

1. No write support yet, so no output_sensor
2. Values read from characteristics are supported as a single-byte int (`uint8_t`) only. Multi-byte values are not supported, including text, integers > 255, or an otherwise binary format like protobuf or packed struct

The author recommends making a new text_sensor platform implementation to take multiple bytes as text, but that still doesn't solve the problem of a binary stream that needs to be unpacked into separate values. It might work to create an esphome internal=true text_sensor that holds the byte array as if it were text - but who knows what kind of havoc that could wreak in esphome.

The more I think of it though, I think the single-number implementation is the "specialized" implementation, and it would be better to convert the sensor implementation to take take the full `vector<uint8>`, and any sensors requiring a single-byte number can just take `x[0]` if that's all they need. But it's been released now, so trying to maintain backwards compatibility with that would surely be problematic :(

closed 2021-06-01 by jesserockz