esphome/feature-requests #558: AC Voltage readings with ESP8266 and zmpt101b

herreran24, 2020-01-09:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

herreran24, 2020-01-09:
I'm trying with the https://surtrtech.com/2019/01/21/easy-measure-of-ac-voltage-using-arduino-and-zmpt101b/ that was made for Arduino, but wasn't able to get to work on the ESP. Regards

olddawgpowers, 2020-07-23:
Did you ever find a solution?

herreran24, 2020-07-24:
No, I did not. I could make it to work in Arduino Nano perfectly then send the readings via serial to the ESP8266. If you need I can send you the code.

olddawgpowers, 2020-07-24:
Thanks for the reply.  I am just looking to sense AC voltage on a rain sensor, I don't really need an accurate measurement.  Hoping maybe something like the zmpt101b would work.

herreran24, 2020-07-24:
The zmpt101b does work, I just wasn't able to make it work on the ESP8266, but it does work on Arduino Nano and it is not as accurate as a digital tester, but it works. A work around is what I did, get a Nano and send the data from it to the ESP, only if you need to send to internet, otherwise, you don't need the ESP, with the nano and a OLED display works perfectly, for example I'm using a SSD1306.

OttoWinter, 2020-07-24:
This bug tracker is for bugs in the ESPHome codebase, not for troubleshooting how to add a certain hardware device.

closed 2020-07-24 by OttoWinter

olddawgpowers, 2020-07-24:
Sorry Otto.  I didn’t realize that.

 

John