esphome/feature-requests #559: Add support for multiple devices with same name (and same code running)

viktorkovarik, 2020-01-10:
**Describe the problem you have/What new integration you would like**

It would be great to add support for devices with same name like by adding random number to host name of some device and then when checking for devices when reflashing having ability to choose the one correct.

**Please describe your use case for this integration and alternatives you've tried:**

I've got multiple light bulbs which I wanted to name to name like "light_bulb" problem is that then I won't be able to determine which is which and possibly I won't be even able to flash it.

I have a lot of bulbs now with the same code and I need to create separate configs for each or rewrite always bulb's name.

**Additional context**

My suggestion would be adding some toggle setting into config file which will append random number (like mybulb_3452) and possibly save it into flash and maybe then change it if there is some device with same name (but that would be quite low probability). Also there would need to be some kind of ability to choose between dns names when OTA flashing. Also maybe would be good to have this number accessible also somehow as display name.

OttoWinter, 2020-01-12:
No, that won't happen.
Much of the codebase operates on the principle that each YAML file corresponds to exactly 1 device, and that each device has some address it can be contacted with.

For example, how should the frontend know which device you want to view logs for when pressing the UPLOAD button.

You can also create a single base YAML file that has all the logic in it, plus one small YAML file each. See FAQ and "substitutions".

closed 2020-01-12 by OttoWinter

viktorkovarik, 2020-01-18:
For the others the best way to have same code on multiple nodes without need to edit multiple files just search for "substitutions:" in esphome documentation. This thing practically does what I wanted.