esphome/feature-requests #560: Add support for VegTrug Grow Care Garden (Mi Flora variant)

ACrazyConcept, 2020-01-11:
**Describe the problem you have/What new integration you would like**

Add support for the plant sensor [VegTrug Grow Care Garden](https://www.vegtrug.com/range/grow-care/grow-care-garden/grow-care-garden/)

**Please describe your use case for this integration and alternatives you've tried:**

It is almost the same as the Mi Flora sensor, just bigger and for outdoor use. The data structure is almost the same and they work with the Flower Care app, but I don't see them with the xiaomi_ble scanner.

**Additional context**

I tried modifying the flora component used in Home Assistant but I can't make it work reliably:
[https://github.com/open-homeautomation/miflora/issues/138](https://github.com/open-homeautomation/miflora/issues/138)
[https://community.home-assistant.io/t/anyone-used-a-plant-sensor-other-than-xaiomi/150868](https://community.home-assistant.io/t/anyone-used-a-plant-sensor-other-than-xaiomi/150868)

So hoping someone in this project can figure it out.

ACrazyConcept, 2020-01-18:
Ammending this
[https://github.com/esphome/esphome/blob/dev/esphome/components/xiaomi_ble/xiaomi_ble.cpp#L80](https://github.com/esphome/esphome/blob/dev/esphome/components/xiaomi_ble/xiaomi_ble.cpp#L80)

`  bool is_hhccjcy01 = (raw[1] & 0x20) == 0x20 && raw[2] == 0x98 && raw[3] == 0x00;`

with:

`bool is_hhccjcy01 = ((raw[1] & 0x20) == 0x20 && raw[2] == 0x98 && raw[3] == 0x00) || ((raw[1] & 0x20) == 0x20 && raw[2] == 0xBC && raw[3] == 0x03);`

I get Moisture, Temperature and Light Intensity.

No conductivity.

A few log examples:

```
[15:04:57][VV][esp32_ble_tracker:286]: Parse Result:
[15:04:57][VV][esp32_ble_tracker:303]:   Address: C4:7C:8D:68:01:B2 (PUBLIC)
[15:04:57][VV][esp32_ble_tracker:305]:   RSSI: -92
[15:04:57][VV][esp32_ble_tracker:306]:   Name: ''
[15:04:57][VV][esp32_ble_tracker:314]:   Ad Flag: 6
[15:04:57][VV][esp32_ble_tracker:317]:   Service UUID: FE:FE95
[15:04:57][VV][esp32_ble_tracker:319]:   Manufacturer data: (0)
[15:04:57][VV][esp32_ble_tracker:320]:   Service data: 71.20.BC.03.16.B2.01.68.8D.7C.C4.0D.08.10.01.3F (16)
[15:04:57][VV][esp32_ble_tracker:323]:   Service Data UUID: FE:FE95
[15:04:57][VV][esp32_ble_tracker:327]: Adv data: 02.01.06.03.02.95.FE.13.16.95.FE.71.20.BC.03.16.B2.01.68.8D.7C.C4.0D.08.10.01.3F (27)
[15:04:57][D][esp32_ble_tracker:456]: Found device C4:7C:8D:68:01:B2 RSSI=-92
[15:04:57][D][esp32_ble_tracker:477]:   Address Type: PUBLIC

[15:05:42][VV][esp32_ble_tracker:286]: Parse Result:
[15:05:42][VV][esp32_ble_tracker:303]:   Address: C4:7C:8D:68:10:B3 (PUBLIC)
[15:05:42][VV][esp32_ble_tracker:305]:   RSSI: -80
[15:05:42][VV][esp32_ble_tracker:306]:   Name: ''
[15:05:42][VV][esp32_ble_tracker:314]:   Ad Flag: 6
[15:05:42][VV][esp32_ble_tracker:317]:   Service UUID: FE:FE95
[15:05:42][VV][esp32_ble_tracker:319]:   Manufacturer data: (0)
[15:05:42][VV][esp32_ble_tracker:320]:   Service data: 71.20.BC.03.1E.B3.10.68.8D.7C.C4.0D.08.10.01.26 (16)
[15:05:42][VV][esp32_ble_tracker:323]:   Service Data UUID: FE:FE95
[15:05:42][VV][esp32_ble_tracker:327]: Adv data: 02.01.06.03.02.95.FE.13.16.95.FE.71.20.BC.03.1E.B3.10.68.8D.7C.C4.0D.08.10.01.26 (27)
```

I think maybe this one below is the conductivity. So not the same as the Mi Flora:

```
[15:05:12][VV][esp32_ble_tracker:286]: Parse Result:
[15:05:12][VV][esp32_ble_tracker:303]:   Address: C4:7C:8D:68:05:0F (PUBLIC)
[15:05:12][VV][esp32_ble_tracker:305]:   RSSI: -80
[15:05:12][VV][esp32_ble_tracker:306]:   Name: ''
[15:05:12][VV][esp32_ble_tracker:314]:   Ad Flag: 6
[15:05:12][VV][esp32_ble_tracker:317]:   Service UUID: FE:FE95
[15:05:12][VV][esp32_ble_tracker:319]:   Manufacturer data: (0)
[15:05:12][VV][esp32_ble_tracker:320]:   Service data: 71.20.BC.03.B4.0F.05.68.8D.7C.C4.0D.0B.10.01.00 (16)
[15:05:12][VV][esp32_ble_tracker:323]:   Service Data UUID: FE:FE95
[15:05:12][VV][esp32_ble_tracker:327]: Adv data: 02.01.06.03.02.95.FE.13.16.95.FE.71.20.BC.03.B4.0F.05.68.8D.7C.C4.0D.0B.10.01.00 (27)
```

ACrazyConcept, 2020-01-19:
I noticed something odd though.

The values I get from an active connection using the official app or a Raspberry Pi, are different from the values recieved on esphome. It seems like the values that are passively broadcast are not refreshed frequently enough inside the device itself.

mauringo, 2020-01-31:
@ACrazyConcept i can help you as soon as i receive the sensors

ACrazyConcept, 2020-02-01:
See my reply on the other repo: [https://github.com/open-homeautomation/miflora/issues/138#issuecomment-581035546](https://github.com/open-homeautomation/miflora/issues/138#issuecomment-581035546)

notelate7942, 2020-04-23:
Hi,

I'm bought 4 x Mi Flora to integrate them with Home assistant and got...4 x Veg Trug...

I tried to integrate them in HA but can't get any value. I tried different workarounds I found on the net, without success.
I see that @ACrazyConcept suggest modifying xiaomi_ble.cpp.
I found the file under 
: /homeassistant/esphome/<name of my ESP>/src/esphome/components/xiaomi_ble/xiaomi_ble.cpp
I tried to modify it as suggested, performed a "clean build" and then "compile" in ESPHOME hoping it would take the changes into account...without success.

see below an extract of my esp yaml file from ESPHOME : 

---
esp32_ble_tracker:

sensor:
  - platform: xiaomi_hhccjcy01
    mac_address: 'C4:7C:8D:67:7B:83'
    temperature:
      name: "Xiaomi HHCCJCY01 Temperature"
    moisture:
      name: "Xiaomi HHCCJCY01 Moisture"
    illuminance:
      name: "Xiaomi HHCCJCY01 Illuminance"
    conductivity:
      name: "Xiaomi HHCCJCY01 Soil Conductivity"
    battery_level:
      name: "Xiaomi HHCCJCY01 Battery Level"
---

all values of the resulting sensor in HA state :"Unknown"

can anyone guide me in the process?

thx in advance!

ACrazyConcept, 2020-04-23:
I am using https://github.com/custom-components/sensor.mitemp_bt now. So I don't know, maybe I should close this issue.
Anyway are you sure it is a 'Garden' sensor?  Check the picture here: https://community.home-assistant.io/t/anyone-used-a-plant-sensor-other-than-xaiomi/150868/18?u=sand

notelate7942, 2020-04-23:
Hi @ACrazyConcept ,

Thx for the swift reply.
Here is a picture of the sensor. 
[image]
I am able to collect the right information from the esp32 with some scripts I run directly on the esp32 but can't make the ESPHome working.
I also tried the mitemp_bt. the issue I had was that the instructions you provided refer to configuration.yaml which I suppose is the way to gowhen the server listens directly through bluetooth..in my/our case, we need to define it in the yaml file of the esp itself, but when I declare : 
- platform: mitemp_bt
... in ESPHOME, it gives an error.
could you please share the configuration you added to configuration.yaml + the yaml of your esp?

thx in advance

ACrazyConcept, 2020-04-23:
That is not the 'Garden' sensor. The one you have work the same as mi flora. No code modification needed. Maybe you have to update its firmware. 
And the mitemp_bt custom component is not for ESPHome, but for home assistant alone.